OPCFoundation/UA-.NETStandard #1145: Client should not verify all endpoint returned by GetEndpoints request

ghord, 2020-10-13:
Opc Ua spec states (emphasis added):

> The Server returns its EndpointDescriptions in the response. Clients use this information to determine whether the list of EndpointDescriptions returned from the DiscoveryEndpoint matches the Endpoints that the Server has. If there is a difference then the Client shall close the Session and report an error. The Server returns all EndpointDescriptions for the serverUri specified by the Client in the request. **The Client only verifies EndpointDescriptions with a transportProfileUri that matches the profileUri specified in the original GetEndpoints request.**

Here in code we can see that client verifies every endpoint it gets, which causes error when connecting to spec compliant server:

https://github.com/OPCFoundation/UA-.NETStandard/blob/9b8cf2eaf2a2e56bdc5bf0070d35c7fc3f6c197b/Libraries/Opc.Ua.Client/Session.cs#L2567-L2575

Client should only verify the endpoints for the requested transport.

ghord, 2020-11-23:
Hi, did you guys had any chance to investigate this issue?
This is blocking the connection to ProSys Simulation Server, which is blocking for us. 
I'm ready to work on and submit PR with a fix if this is something that could be merged.

AlinMoldovean, 2020-11-23:
Hello @ghord,

I had a look at this.
The `Session` [constructor ](https://github.com/OPCFoundation/UA-.NETStandard/blob/34d0af8dd64db01340966511244a98e7e2e2ff5b/Libraries/Opc.Ua.Client/Session.cs#L97-L102) should be extended with a `profileUris` parameter to allow the application to send the _profileUris_ value used at `GetEndpoints()`.
The validation logic at the place you pointed should be updated to also take in use the `profileUris`.
I will put a fix soon for this.

This should not be a blocking issue for you if you use the connect example from _ConsoleReferenceClient_ project [here](https://github.com/OPCFoundation/UA-.NETStandard/blob/34d0af8dd64db01340966511244a98e7e2e2ff5b/Applications/ConsoleReferenceClient/UAClient.cs#L98-L107) because Session.Create() does not send `availableEndpoints` for validation.
Or you can send `availableEndpoints = null` if you use this Session [constructor](https://github.com/OPCFoundation/UA-.NETStandard/blob/34d0af8dd64db01340966511244a98e7e2e2ff5b/Libraries/Opc.Ua.Client/Session.cs#L97-L102)

ghord, 2020-11-24:
Hello @AlinMoldovean 

Thank you for looking into the issue and your prompt response.

Unfortunately disabling `availableEndpoints` check altogether will mean that our client is no longer compliant with spec as it will no longer verify any endpoints. 
Since you mentioned you are already working on a fix, we decided to wait and upgrade once it is available.

AlinMoldovean, 2020-11-25:
Hi @ghord,

Could you please test with your server  if #1194 fixes the issue?

ghord, 2020-11-26:
Hi @AlinMoldovean 
Than you for quick fix.
I'll test the fix and get back to you.

ghord, 2020-11-30:
Hi @AlinMoldovean 
I've tested the PR version of the library and it now successfully connects to Prosys simulation server once I pass `EndpointDescription.TransportProfileUri` to Session constructor.

PR #1194 (2020-12-02): Fix for validation of endpoints from GetEndpoints and  CreateSession responses.
Closes #1145
commit 7f19479b (2020-12-02): Closes #1145 - Fix for validation of endpoints from GetEndpoints and CreateSession responses.
closed 2020-12-02 by AlinMoldovean