OPCFoundation/UA-.NETStandard #1152: Unable to accept invalid server certificate

lkomanetz, 2020-10-15:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [ ] Question [x] Help wanted

**Describe the Issue**

When creating an OPC session the server certificate is invalid for one reason or another.  In this case it is the default certificate that is valid from 2004 - 2007.  The `CertificateValidator.CertificateValidation` event handler is never called and I don't seem to have a way to ignore the invalid certificate from the OPC server.  I am able to connect to the server through OPC Expert just by giving it an IP address.  OPC Expert does not warn me or have any issues browsing or observing PLC tags with the same server that my .NET Core app (using the .NET Standard OPC UA NUGET package) can't establish a connection to.

**To Reproduce**
Steps to reproduce the behavior:
1. Have an OPC server that issues a certificate that is already expired.
2. Attempt to establish a connection to the OPC server with the .NET Standard library `Session.Create(...)`

**Expected behavior**
I have the ability to decide if I want to connect to the server anyway regardless of the certificate.

**Log output**
```

************************* Logging started at 10/13/2020 17:32:26
10/13/2020 17:32:26.560 Checking application instance certificate.
10/13/2020 17:32:26.579 Checking application instance certificate. CN=Twin Axis, C=US, S=Arizona, O=OPC Foundation, DC=win-16ojh701kcc
10/13/2020 17:32:27.637 Channel 0 in Connecting state.
10/13/2020 17:32:27.837 Channel 0: SendHelloMessage()
10/13/2020 17:32:27.873 Channel 0: ProcessAcknowledgeMessage()
10/13/2020 17:32:27.888 Channel 0 in Opening state.
10/13/2020 17:32:27.908 Token #0 created. CreatedAt = 22:32:27.907 . Lifetime = 3600000
10/13/2020 17:32:27.959 Channel 0: ProcessOpenSecureChannelResponse()
10/13/2020 17:32:27.988 Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
10/13/2020 17:32:28.002 Sender Certificate: (none)
10/13/2020 17:32:28.041 SECURE CHANNEL CREATED [.NetStandard ClientChannel UA-TCP 1.04.361.39] [ID=1172064] Connected To: opc.tcp://192.168.1.120:4840/ [None/None/Binary]
10/13/2020 17:32:28.058 Token #1 activated. CreatedAt = 22:32:27.907 . Lifetime = 3600000
10/13/2020 17:32:28.072 Channel 1172064 in Open state.
10/13/2020 17:32:28.139 Channel 1172064: Close
10/13/2020 17:32:28.153 Channel 1172064 in Closing state.
10/13/2020 17:32:28.169 Channel 1172064: SendCloseSecureChannelRequest()
10/13/2020 17:32:28.197 Channel 0 in Closed state.
10/13/2020 17:32:28.212 CLIENTCHANNEL SOCKET CLOSED: 03B95145, ChannelId=1172064
10/13/2020 17:32:28.339 ***EXCEPTION*** BadCertificateTimeInvalid 'Certificate has is expired or not yet valid. NotTimeValid: A required certificate is not within its validity period when verifying against the current system clock or the timestamp in the signed file.'
10/13/2020 17:32:28.357 Certificate Vaildation failed for 'CN=TASS_000_000_001, OU=Intelligent Platforms, O=GE, L="Charlottesville, Va.", C=US, DC=vxTarget'. Reason=BadCertificateTimeInvalid
10/13/2020 17:32:28.372 Certificate 'CN=TASS_000_000_001, OU=Intelligent Platforms, O=GE, L="Charlottesville, Va.", C=US, DC=vxTarget' rejected. Reason=BadCertificateTimeInvalid
10/13/2020 17:32:28.389 Writing rejected certificate to directory: [Directory]/storage/emulated/0/OPC Foundation/PKI/rejected
10/13/2020 17:32:28.647 ***EXCEPTION*** BadCertificateTimeInvalid ServiceResultException Certificate has is expired or not yet valid. NotTimeValid: A required certificate is not within its validity period when verifying against the current system clock or the timestamp in the signed file.
```

**Code snippets**
```cs
public async Task OpenAsync() {
	var opcUrl = $"opc.tcp://{IpAddress}:{Port}";
	var appConfig = await LoadOpcConfigurationAsync().ConfigureAwait(false);
	await appConfig.Validate(ApplicationType.Client);
	var endpoint = ConfigureEndpoint(opcUrl, appConfig);
	_session = await Session.Create(
		configuration: appConfig,
		endpoint: endpoint,
		updateBeforeConnect: false,
		sessionName: GetSessionName(),
		sessionTimeout: 60_000,
		identity: null,
		preferredLocales: null
	);

	static ConfiguredEndpoint ConfigureEndpoint(string url, ApplicationConfiguration appConfig) {
		var selectedEndpoint = OpcClientUtils.SelectEndpoint(url, false, 15_000);
		var endpointConfig = EndpointConfiguration.Create(appConfig);
		var endpoint = new ConfiguredEndpoint(selectedEndpoint.Server, endpointConfig);
		endpoint.Update(selectedEndpoint);
		return endpoint;
	}

	static string GetSessionName() {
[... 111 lines cut ...]
        <!-- <TraceMasks>515</TraceMasks> -->
        <!-- Show Only Security, COM Calls, Errors and Trace -->
        <!-- <TraceMasks>771</TraceMasks> -->
        <!-- Show Only Security, Service Calls, Errors and Trace -->
        <!-- <TraceMasks>523</TraceMasks> -->
        <!-- Show Only Security, ServiceResultExceptions, Errors and Trace -->
        <!--<TraceMasks>519</TraceMasks>-->
    </TraceConfiguration>

    <!-- Disables the hi-res clock if the QueryPerformanceCounter does work on a particular machine. -->
    <DisableHiResClock>true</DisableHiResClock>

</ApplicationConfiguration>
```
**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.7.2
 - Runtime: .NET Core 3.1
 - OPC Package Release Version 1.4.361.39
 - Component: Opc.Ua.Client
 - Server: Built into Emerson PLC
 - Client: I'm building the client

**Additional context**
We do plan on using certificates and will need to figure out how to issue a new certificate to the server but for now we just need this to connect so they can test the machine.

mregen, 2020-10-16:
Hi @lkomanetz,

there should be two ways to solve this:
i) in the CertificateValidation callback set `Accept` to true
( from the log I can see the callback should be called if it is configured correctly..)

ii)
CertificateValidator can suppress expired cert errors:
` ((id.ValidationOptions & CertificateValidationOptions.SuppressCertificateExpired)`

could you try this to configure (rejected cert must be moved to trusted store):

```
<TrustedPeerCertificates>
    <StoreType>Directory</StoreType>
    <StorePath>./opc/PKI/trusted</StorePath>
    <ValidationOptions>1</ValidationOptions>
</TrustedPeerCertificates>
```

lkomanetz, 2020-10-16:
@mregen I work remote so I'm waiting for the credentials to get on the computer that has the software running and can connect to the OPC server built into the PLC.  In the above code I have put log statements in the event handler and have not seen it fired (even after removing the `hasCertificate` condition).  Does the `ValidationOptions` XML tag ensure that the event gets fired for trusted peer certificates?

mregen, 2020-10-17:
Hi @lkomanetz, from your personal message I see that this issue is resolved, please close if appropriate...

closed 2020-10-17 by lkomanetz