OPCFoundation/UA-.NETStandard #1153: Could not load configuration file in Unity

soaibsafi, 2020-10-16:
I am trying to implement OPC UA client in Unity. But I am getting an `ServiceResultException: Could not load configuration file.` expection in line `application.LoadApplicationConfiguration("ConsoleReferenceClient.Config.xml", false).Wait();`
which means the "Config.xml" file couldn't be loaded in Unity. But the same program works perfectly in console application. **How can I load the configuration file in unity?**
```
void Start()
{
        Debug.Log("OPC UA Console Reference Client");

        // Define the UA Client application
        ApplicationInstance application = new ApplicationInstance();
            application.ApplicationName = "Quickstart Console Reference Client";
            application.ApplicationType = ApplicationType.Client;
        
        // load the application configuration.
        application.LoadApplicationConfiguration("Assets/ConsoleReferenceClient.Config.xml", false).Wait();
            // check the application certificate.
            application.CheckApplicationInstanceCertificate(false, 0).Wait();

            // create the UA Client object and connect to configured server.
            Quickstarts.ConsoleReferenceClient.UAClient uaClient = new Quickstarts.ConsoleReferenceClient.UAClient(application.ApplicationConfiguration);

            if (uaClient.Connect())
            {
                // Run tests for available methods.
                uaClient.ReadNodes();
                //uaClient.WriteNodes();
                uaClient.Browse();
                //uaClient.CallMethod();

                uaClient.SubscribeToDataChanges();
                // Wait for some DataChange notifications from MonitoredItems
                System.Threading.Thread.Sleep(20000);

                uaClient.Disconnect();
            }
            else
            {
                Debug.Log("Could not connect to server!");
            }

            Debug.Log("\nProgram ended.");
            Debug.Log("Press any key to finish...");
            Console.ReadKey();
}
```

**Exception**:
```
ServiceResultException: Could not load configuration file.
Opc.Ua.Configuration.ApplicationInstance+<LoadApplicationConfiguration>d__33.MoveNext () (at <1c5dac388c934c65a66a1f12b206c16d>:0)
Rethrow as AggregateException: One or more errors occurred.
System.Threading.Tasks.Task.ThrowIfExceptional (System.Boolean includeTaskCanceledExceptions) (at <fb001e01371b4adca20013e0ac763896>:0)
System.Threading.Tasks.Task.Wait (System.Int32 millisecondsTimeout, System.Threading.CancellationToken cancellationToken) (at <fb001e01371b4adca20013e0ac763896>:0)
System.Threading.Tasks.Task.Wait () (at <fb001e01371b4adca20013e0ac763896>:0)
TestClient.Start () (at Assets/Scripts/TestClient.cs:22)
```

AlinMoldovean, 2020-10-16:
Hi @soaibsafi,

You should debug inside the `LoadApplicationConfiguration()` method and find the exact place where the exception is generated and also the reason.
Another option is to define the configuration programmatically without loading it from the xml config file.
Like this:
```
                ApplicationConfiguration configuration = new ApplicationConfiguration();

                configuration.ApplicationName = "Quickstart Console Reference Client";
                configuration.ApplicationType = ApplicationType.Client;
                configuration.ApplicationUri = $"urn:{Utils.GetHostName()}:Quickstarts:Console ReferenceClient";
                configuration.TransportConfigurations = new TransportConfigurationCollection();
                configuration.TransportQuotas = new TransportQuotas { OperationTimeout = 15000 };
                configuration.ClientConfiguration = new ClientConfiguration { DefaultSessionTimeout = 60000 };

                configuration.SecurityConfiguration = new SecurityConfiguration {
                    ApplicationCertificate = new CertificateIdentifier {
                        StoreType = CertificateStoreType.Directory,
                        StorePath = @"%CommonApplicationData%\OPC Foundation\pki\own",
                        SubjectName = "Quickstart Console Reference Client"
                    },
                    TrustedPeerCertificates = new CertificateTrustList {
                        StoreType = CertificateStoreType.Directory,
                        StorePath = @"%CommonApplicationData%\OPC Foundation\pki\trusted",
                    },
                    TrustedIssuerCertificates = new CertificateTrustList {
                        StoreType = CertificateStoreType.Directory,
                        StorePath = @"%CommonApplicationData%\OPC Foundation\pki\issuer",
                    },
                    RejectedCertificateStore = new CertificateTrustList {
                        StoreType = CertificateStoreType.Directory,
                        StorePath = @"%CommonApplicationData%\OPC Foundation\pki\rejected",
                    },
                    AutoAcceptUntrustedCertificates = true
                };

                configuration.TraceConfiguration = new TraceConfiguration() {
                    OutputFilePath = @"%CommonApplicationData%\OPC Foundation\Logs\Quickstarts.ConsoleReferenceClient.log.txt",
                    TraceMasks = 1
                };

                configuration.Validate(ApplicationType.Client).Wait();
                application.ApplicationConfiguration = configuration;

                // check the application certificate.
                application.CheckApplicationInstanceCertificate(false, 0).Wait();

```

soaibsafi, 2020-10-23:
@AlinMoldovean,
Thank you for your suggestions. I have tried the both way. The problem is not with loading the Configuration file. Some of my assembly couldn't be loaded like `Error: Could not load signature of Microsoft.Extensions.Primitives.CancellationChangeToken`.
 But I am not sure about how can I resolve it. I have [created an issue at .Net/sdk](https://github.com/dotnet/sdk/issues/14298) regarding that. 
Please let me know If you can suggest something for the possible solution.

soaibsafi, 2020-10-23:
@AlinMoldovean,
Thanks for your suggestion.
I have solved the problem. The problem was not with the **Configuration file**. The problem was with the assembly which I have imported into unity. The assemblies was broken and I was getting **Could not load signature**  error.  I would like to post the solution in case someone might find this helpful in the future.

 The problem was that I have added the dlls by downloading through NuGet and added them in Plugins folder inside Unity project. That probably created some dependencies that they were working inside the editor but the dll was unable to find the corresponding package when building the app. That's why I was getting the error.

The solution came by using [NuGetForUnity](https://github.com/GlitchEnzo/NuGetForUnity/releases/tag/v2.0.1) that automatically downloads NuGet packages and translates them to unity packages.

I hope this answer may come in handy.

closed 2020-10-23 by soaibsafi