OPCFoundation/UA-.NETStandard #1155: Question: is this a possible typo, that could lead to a bug?

rasane, 2020-10-16:
ref: https://github.com/OPCFoundation/UA-.NETStandard/blob/7f3f0dbbb4f75f141115a9371301f0cf59c076a3/Stack/Opc.Ua.Core/Types/BuiltIn/SessionLessServiceMessage.cs#L60

Hi, 
apologies if I am asking this prematurely. At this moment, I am learning, simply reading the standard and then the code to get a better understanding.
The loop referenced above is encoding 'ServerUris', the destination size is for that. Yet the limits enforced in the loop appears to be NamespaceUris.Count, shouldn't it be ServerUris.Count?
apologies if this is silly and thanks in advance, rasane

mregen, 2020-10-17:
Thanks @rasane, good catch --> thats a bug

rasane, 2020-10-19:
Hi, happy to raise a PR if okay with you..
question
- I can do a string change and raise a trivial PR - is this sufficient?
- I spent some time trying to write a failing test and a fix test for the item.. it is pretty close but not done.. concerned about keeping the bug open too long, so happy to slice the first one in and then follow up with the test, if that is acceptable..

let me know.. 
thanks, rasane

mregen, 2020-10-20:
yes please, thanks for raising a PR. And having more tests is even more welcome, anytime!

Cheers,
Martin

rasane, 2020-10-21:
Hi, there is 1 failed job in the build called "release test on netcoreapp 2.2" which has failed across the pipelines several other times.. should I be "rerunning this job" as an attempt to fix or is there any other suggestion? 

reading through the logs: 
There are a couple of warnings about "_SDK/runtime 2.2 and 3.0 are now End of Life(EOL)_ " and  "_Info: Azure Pipelines hosted agents have been updated to contain .Net Core 3.x (3.1) SDK/Runtime along with 2.1. Unless you have locked down a SDK version for your project(s), 3.x SDK might be picked up which might have breaking behavior as compared to previous versions._ ".. 

All of this may be unrelated, but just checking if 
a) this build failures is an issue for / due to this PR
b) do I need to do something / I have no idea where to begin.. 

thanks, rasane

commit 03eed2d9 (2020-10-21): Added tests for the Sessionlessmessage #1155 (#1159)

fixes #1155
closed 2020-10-21 by mregen