OPCFoundation/UA-.NETStandard #1158: Loading the MachineTool Nodeset

ebrithiljonas, 2020-10-20:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [ ] Question [X] Help wanted

**Describe the Issue**

I'm trying to load in the MachineTool Nodeset from the official UA-Nodeset repository. But I always get this Error:

```csharp
Exception has occurred: CLR/Opc.Ua.ServiceResultException
An exception of type 'Opc.Ua.ServiceResultException' occurred in Opc.Ua.Core.dll but was not handled in user code: 'A valid supertype identifier is required.'
   at Opc.Ua.TypeTable.AddSubtype(NodeId subTypeId, NodeId superTypeId, QualifiedName browseName)
   at Opc.Ua.Server.CustomNodeManager2.AddPredefinedNode(ISystemContext context, NodeState node)
   at Demo.DemoNodeManager.ImportXml(IDictionary`2 externalReferences, String resourcepath) in /home/****/UA_dotNETStandard/Demo/DemoNodeManager.cs:line 122
   at Demo.DemoNodeManager.CreateAddressSpace(IDictionary`2 externalReferences) in /home/****/UA_dotNETStandard/Demo/DemoNodeManager.cs:line 102
   at Opc.Ua.Server.MasterNodeManager.Startup()
```

**To Reproduce**

Add the NodeManager to the SampleServer and run it:

```csharp
nodeManagers.Add(new global::Demo.DemoNodeManager(server, configuration));
```

**Expected behavior**

Load the MachineTool Nodes into the server.

**Code snippets**

This is the NodeManager I'm using.

```csharp
/* ========================================================================
 * Copyright (c) 2005-2016 The OPC Foundation, Inc. All rights reserved.
 *
 * OPC Foundation MIT License 1.00
 * 
 * Permission is hereby granted, free of charge, to any person
 * obtaining a copy of this software and associated documentation
 * files (the "Software"), to deal in the Software without
 * restriction, including without limitation the rights to use,
 * copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the
 * Software is furnished to do so, subject to the following
 * conditions:
 * 
 * The above copyright notice and this permission notice shall be
 * included in all copies or substantial portions of the Software.
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
 * EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
 * OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
 * NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
 * HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
 * WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 * FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
 * OTHER DEALINGS IN THE SOFTWARE.
 *
 * The complete license agreement can be found here:
 * http://opcfoundation.org/License/MIT/1.00/
 * ======================================================================*/

using System;
using System.Collections.Generic;
using System.Xml;
using System.Threading;
using Opc.Ua;
using Opc.Ua.Server;
using System.IO;
using System.Reflection;

namespace Demo
{
    /// <summary>
    /// A node manager the diagnostic information exposed by the server.
    /// </summary>
    public class DemoNodeManager : Opc.Ua.Server.CustomNodeManager2
    {
        #region Constructors
        /// <summary>
        /// Initializes the node manager.
        /// </summary>
        public DemoNodeManager(IServerInternal server, ApplicationConfiguration configuration)
        :
            base(server, configuration, Namespaces.Demo)
        {
            SystemContext.NodeIdFactory = this;

            m_configuration = configuration.ParseExtension<ReferenceServerConfiguration>();

            if (m_configuration == null)
            {
                m_configuration = new ReferenceServerConfiguration();
            }

        }
        #endregion

        #region INodeIdFactory Members
        /// <summary>
        /// Creates the NodeId for the specified node.
        /// </summary>
        /// <param name="context">The context.</param>
        /// <param name="node">The node.</param>
        /// <returns>The new NodeId.</returns>
        public override NodeId New(ISystemContext context, NodeState node)
        {
            uint id = Utils.IncrementIdentifier(ref m_lastUsedId);
            return new NodeId(id, m_namespaceIndex);
        }
        #endregion

        #region INodeManager Members
        /// <summary>
        /// Does any initialization required before the address space can be used.
        /// </summary>
        /// <remarks>
        /// The externalReferences is an out parameter that allows the node manager to link to nodes
        /// in other node managers. For example, the 'Objects' node is managed by the CoreNodeManager and
        /// should have a reference to the root folder node(s) exposed by this node manager.  
        /// </remarks>
        public override void CreateAddressSpace(IDictionary<NodeId, IList<IReference>> externalReferences)
        {
            lock (Lock)
            {
                IList<IReference> references = null;

                if (!externalReferences.TryGetValue(ObjectIds.ObjectsFolder, out references))
                {
                    externalReferences[ObjectIds.ObjectsFolder] = references = new List<IReference>();
                }

                string resourcepath = "/home/****/Opc.Ua.MachineTool.NodeSet2.xml"; 
                ImportXml(externalReferences, resourcepath);                
            }
        }

        /// <summary>
        /// Import NodeSets from xml
        /// </summary>
        /// <param name="path">String to path of XML</param>
        private void ImportXml(IDictionary<NodeId, IList<IReference>> externalReferences, string resourcepath)
        {
            NodeStateCollection predefinedNodes = new NodeStateCollection();

            Stream stream = new FileStream(resourcepath, FileMode.Open);
            Opc.Ua.Export.UANodeSet nodeSet = Opc.Ua.Export.UANodeSet.Read(stream);

            SystemContext.NamespaceUris.Append(nodeSet.NamespaceUris.ToString());
            nodeSet.Import(SystemContext, predefinedNodes);

            for (int ii = 0; ii < predefinedNodes.Count; ii++)
            {
                AddPredefinedNode(SystemContext, predefinedNodes[ii]);
            }
            // ensure the reverse refernces exist.
            AddReverseReferences(externalReferences);
        }
        /// <summary>
        /// Replaces the generic node with a node specific to the model.
        /// </summary>
        protected override NodeState AddBehaviourToPredefinedNode(ISystemContext context, NodeState predefinedNode)
        {
            BaseObjectState passiveNode = predefinedNode as BaseObjectState;

            if (passiveNode == null)
            {
                return predefinedNode;
            }

            NodeId typeId = passiveNode.TypeDefinitionId;

            if (!IsNodeIdInNamespace(typeId) || typeId.IdType != IdType.Numeric)
            {
                return predefinedNode;
            }
            return predefinedNode;
        }

        #endregion

        #region Private Fields
        private ReferenceServerConfiguration m_configuration;
        private List<BaseDataVariableState> m_dynamicNodes;
        private ushort m_namespaceIndex;
        private ushort m_typeNamespaceIndex;
        private long m_lastUsedId;
        #endregion
    }
}
```

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Development environment: Visual Studio Code
 - Runtime: .Net Core 3.1
 - OPC Package Release Version: 1.4.363.107
 - Server: NetCoreConsoleServer
 - Client: UA Expert

AlinMoldovean, 2020-10-20:
Hi @ebrithiljonas ,

"_Opc.Ua.MachineTool.NodeSet2.xm_l" has some NodeSet2.xml dependencies that should be first loaded in the correct order.

https://github.com/OPCFoundation/UA-Nodeset/blob/393b633468a5d1d062dd253e1488d1d8ba335b6f/MachineTool/Opc.Ua.MachineTool.NodeSet2.xml#L38-L44

ebrithiljonas, 2020-10-20:
Okey, I tried that, first only with the Machinery Nodeset. I can load the required /UA/DI/ Nodeset from the _Opc.Ua.Di.PredefinedNodes.uanodes_ File without any problems. But when I try to load the _Opc.Ua.Machinery.NodeSet2.xml_ it always gives me this Error:
```csharp
Exception has occurred: CLR/Opc.Ua.ServiceResultException
An exception of type 'Opc.Ua.ServiceResultException' occurred in Opc.Ua.Core.dll but was not handled in user code: 'Could not resolve namespace uri: http://opcfoundation.org/UA/2011/03/UANodeSet.xsd'
   at Opc.Ua.NodeState.LoadNode(ISystemContext context, XmlDecoder decoder)
   at Opc.Ua.NodeStateCollection.LoadFromXml(ISystemContext context, Stream istrm, Boolean updateTables)
   at Umati.MachineryNodeManager.LoadPredefinedNodes(ISystemContext context) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Umati/MachineryNodeManager.cs:line 76
   at Opc.Ua.Sample.SampleNodeManager.LoadPredefinedNodes(ISystemContext context, IDictionary`2 externalReferences) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Base/SampleNodeManager.cs:line 391
   at Opc.Ua.Sample.SampleNodeManager.CreateAddressSpace(IDictionary`2 externalReferences) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Base/SampleNodeManager.cs:line 347
   at Umati.MachineryNodeManager.CreateAddressSpace(IDictionary`2 externalReferences) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Umati/MachineryNodeManager.cs:line 64
   at Opc.Ua.Server.MasterNodeManager.Startup()
```

I tried to adapt the Nodemanager from the "Boiler" example and it looks like this:

```csharp

using System.Collections.Generic;
using Opc.Ua;
using Opc.Ua.Sample;
using System.IO;
using System.Reflection;

namespace Umati
{
    public class MachineryNodeManager : SampleNodeManager
    {
        #region Private Fields
        private ushort m_namespaceIndex;
        private ushort m_typeNamespaceIndex;
        private long m_lastUsedId;
        #endregion

        /// <summary>
        /// Initializes the node manager.
        /// </summary>
        public MachineryNodeManager(
            Opc.Ua.Server.IServerInternal server, 
            ApplicationConfiguration configuration)
        :
            base(server)
        {
            List<string> namespaceUris = new List<string>();
            namespaceUris.Add("http://opcfoundation.org/UA/Machinery/");
            namespaceUris.Add("http://opcfoundation.org/UA/Machinery/Instance");
            NamespaceUris = namespaceUris;

            m_typeNamespaceIndex = Server.NamespaceUris.GetIndexOrAppend(namespaceUris[0]);
            m_namespaceIndex = Server.NamespaceUris.GetIndexOrAppend(namespaceUris[1]);

            m_lastUsedId = 0;
        }

        #region INodeIdFactory Members
        /// <summary>
        /// Creates the NodeId for the specified node.
        /// </summary>
        /// <param name="context">The context.</param>
        /// <param name="node">The node.</param>
        /// <returns>The new NodeId.</returns>
        public override NodeId New(ISystemContext context, NodeState node)
        {
            uint id = Utils.IncrementIdentifier(ref m_lastUsedId);
            return new NodeId(id, m_namespaceIndex);
        }
        #endregion

        /// <summary>
        /// Loads a node set from a file or resource and addes them to the set of predefined nodes.
        /// </summary>
        protected override NodeStateCollection LoadPredefinedNodes(ISystemContext context)
        {
            NodeStateCollection predefinedNodes = new NodeStateCollection();
            Stream stream = new FileStream("Nodeset/Machinery/Opc.Ua.Machinery.NodeSet2.xml", FileMode.Open);
            predefinedNodes.LoadFromXml(context, stream, true);
            return predefinedNodes;
        }

    }

}
```

ebrithiljonas, 2020-10-21:
I got it to a state where I can load the _Opc.Ua.Machinery.NodeSet2.xml_ without any problem. But when I want to load the next Nodeset it throws this error:

```csharp
Exception has occurred: CLR/Opc.Ua.ServiceResultException
An exception of type 'Opc.Ua.ServiceResultException' occurred in Opc.Ua.Core.dll but was not handled in user code: 'A valid supertype identifier is required.'
   at Opc.Ua.TypeTable.AddSubtype(NodeId subTypeId, NodeId superTypeId, QualifiedName browseName)
   at Opc.Ua.Sample.SampleNodeManager.AddTypesToTypeTree(BaseTypeState type) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Base/SampleNodeManager.cs:line 693
   at Opc.Ua.Sample.SampleNodeManager.AddPredefinedNode(ISystemContext context, NodeState node) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Base/SampleNodeManager.cs:line 430
   at Umati.IANodeManager.LoadPredefinedNodes(ISystemContext context) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Umati/IANodeManager.cs:line 87
   at Opc.Ua.Sample.SampleNodeManager.LoadPredefinedNodes(ISystemContext context, IDictionary`2 externalReferences) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Base/SampleNodeManager.cs:line 391
   at Opc.Ua.Sample.SampleNodeManager.CreateAddressSpace(IDictionary`2 externalReferences) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Base/SampleNodeManager.cs:line 347
   at Umati.IANodeManager.CreateAddressSpace(IDictionary`2 externalReferences) in /home/jonas/Documents/Retrokit/UA_dotNETStandard/Umati/IANodeManager.cs:line 65
   at Opc.Ua.Server.MasterNodeManager.Startup()
```
This is the way I load the Nodes now, for Machinery and IA Nodeset:

```csharp

using System.Collections.Generic;
using Opc.Ua;
using Opc.Ua.Sample;
using System.IO;
using System.Reflection;
using Opc.Ua.Export;

namespace Umati
{
    public class MachineryNodeManager : SampleNodeManager
    {
        #region Private Fields
        private ushort m_namespaceIndex;
        private ushort m_typeNamespaceIndex;
        private long m_lastUsedId;
        #endregion

        /// <summary>
        /// Initializes the node manager.
        /// </summary>
        public MachineryNodeManager(
            Opc.Ua.Server.IServerInternal server, 
            ApplicationConfiguration configuration)
        :
            base(server)
        {
            List<string> namespaceUris = new List<string>();
            namespaceUris.Add("http://opcfoundation.org/UA/Machinery/");
            namespaceUris.Add("http://opcfoundation.org/UA/Machinery/Instance");
            NamespaceUris = namespaceUris;

            m_typeNamespaceIndex = Server.NamespaceUris.GetIndexOrAppend(namespaceUris[0]);
            m_namespaceIndex = Server.NamespaceUris.GetIndexOrAppend(namespaceUris[1]);

            m_lastUsedId = 0;
        }

        #region INodeIdFactory Members
        /// <summary>
        /// Creates the NodeId for the specified node.
        /// </summary>
        /// <param name="context">The context.</param>
        /// <param name="node">The node.</param>
        /// <returns>The new NodeId.</returns>
        public override NodeId New(ISystemContext context, NodeState node)
        {
            uint id = Utils.IncrementIdentifier(ref m_lastUsedId);
            return new NodeId(id, m_namespaceIndex);
        }
        #endregion

        /// <summary>
        /// Does any initialization required before the address space can be used.
        /// </summary>
        /// <remarks>
        /// The externalReferences is an out parameter that allows the node manager to link to nodes
        /// in other node managers. For example, the 'Objects' node is managed by the CoreNodeManager and
        /// should have a reference to the root folder node(s) exposed by this node manager.  
        /// </remarks>
        public override void CreateAddressSpace(IDictionary<NodeId, IList<IReference>> externalReferences)
        {
            lock (Lock)
            {
                base.CreateAddressSpace(externalReferences);
                // CreateBoiler(SystemContext, 2);
            }
        }

        /// <summary>
        /// Loads a node set from a file or resource and addes them to the set of predefined nodes.
        /// </summary>
        protected override NodeStateCollection LoadPredefinedNodes(ISystemContext context)
        {
            NodeStateCollection predefinedNodes = new NodeStateCollection();
            Stream stream = new FileStream("Nodeset/Machinery/Opc.Ua.Machinery.NodeSet2.xml", FileMode.Open);
            UANodeSet nodeSet = UANodeSet.Read(stream);

            foreach (string namespaceUri in nodeSet.NamespaceUris)
            {
                context.NamespaceUris.GetIndexOrAppend(namespaceUri);
            }
            nodeSet.Import(context, predefinedNodes);

            for (int ii = 0; ii < predefinedNodes.Count; ii++)
            {
                AddPredefinedNode(SystemContext, predefinedNodes[ii]);
            }

            return predefinedNodes;
        }

    }

}
```

STUBA-rupr, 2020-12-28:
I have similar issue with Administrative Asset Shell Nodeset2.xml.
The `nodeSet.Import(context, predefinedNodes);`
does not set namespace index in `CustomNodeManger.Server.TypeTree` key.

mrsuciu, 2021-09-10:
Since the CustomTypes are not supported yet for servers, the node-sets which contain such types may have issues to be loaded.
We are planning to support NodeSet2.xml import.

closed 2021-09-10 by mrsuciu

vj300, 2025-02-02:
For anyone out there struggling with this issue, I had the same error 'A valid supertype identifier is required.'
I was able to resolve this by sorting the Nodeset2.xml files with the parent nodes coming before any of its child nodes.