OPCFoundation/UA-.NETStandard #1160: Request to submit a PR

lkomanetz, 2020-10-20:
**Type of Issue**
[x] Bug [ ] Enhancement [ ] Compliance [ ] Question [x] Help wanted

I discovered an issue with `ApplicationConfiguration` and how it re-assigns the `CertificateValidator` instance when `Validate(...)` is called.  I pulled the code and want to submit a PR but cannot push to the repo.  Is that something the community can do or is that not allowed?

mregen, 2020-10-20:
You can only push to a fork, then open the PR from your fork.

lkomanetz, 2020-10-20:
That makes perfect sense.  I've been so used to working with GIT within my organization that I forgot about using forks internally at my job that I forgot about proper open source etiquette.

closed 2020-10-20 by lkomanetz