OPCFoundation/UA-.NETStandard #1166: Cannot access certificate private key, but I am using AnonymousUserTokenPolicy

Havlat82, 2020-10-22:
OPC UA Server fail to start due exception Cannot access certificate private key, but I am using AnonymousUserTokenPolicy
I really do not understand why do get this exception. It all worked yesterday evening. I did not change config file and I checked it is same. I did not change code either. Just run the app and this happen. I am confused where is problem. Please, can anyone help me?

`<UserTokenPolicies>

      
      <ua:UserTokenPolicy>
        <ua:TokenType>Anonymous_0</ua:TokenType>
        <ua:SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#None</ua:SecurityPolicyUri>
      </ua:UserTokenPolicy>     

      
      
      
    
    </UserTokenPolicies>`

I am starting server with this method:
`private void InitServer()
        {
            ApplicationInstance app = new ApplicationInstance
            {
                ApplicationName = "RoboBar.TestServer",
                ApplicationType = ApplicationType.Server,
                ConfigSectionName = "RoboBar.TestServer"
            };

            try
            {
                var configuration = app.LoadApplicationConfiguration(false);
                var certificate = app.CheckApplicationInstanceCertificate(false, 0);
                certificate.Wait();
                var certOK = certificate.Result;
                if (!certOK)
                {
                    throw new Exception("Application instance certificate invalid!");
                }

                var startServer = app.Start(new RoboBarServer());
                startServer.Wait();
            }
            catch (Exception ex)
            {
                string text = "Exception: " + ex.Message;
                if (ex.InnerException != null)
                {
                    text += "\r\nInner exception: ";
                    text += ex.InnerException.Message;
                }
                MessageBox.Show(text, app.ApplicationName);
            }

            Server = app.Server as StandardServer;
            Configuration = app.ApplicationConfiguration;
        }`
It is run in BackgroundWorker.. 
Please can anyone tell me, where is a problem?
Thanks in advance. Petr

Zigby, 2020-11-06:
Application Instance Certificates are used regardless of UserTokenPolicy to authenticate UA Applications (and establish trust between applications) The xml configuration file should contain an <ApplicationCertificate>section where the location of the certificate is specified. Make sure the user account running the server has access to the specified folder (in case store type is directory). The SDK will create a self-signed certificate if it cannot be found.
Here is how it is specified in the sample Reference Server
```
<SecurityConfiguration>
  <!-- Where the application instance certificate is stored (MachineDefault) -->
  <ApplicationCertificate>
    <StoreType>Directory</StoreType>
    <StorePath>%LocalApplicationData%/OPC Foundation/pki/own</StorePath>
    <SubjectName>CN=Quickstart Reference Server, C=US, S=Arizona, O=OPC Foundation, DC=localhost</SubjectName>
  </ApplicationCertificate>
```

The certificates are also used to derive keys used for signing and encryption depending on the Security Policy.
I assume using the security Policy None should not require an application instance certificate, but the others definitely needs one.

````
<SecurityPolicies>
...
  <SecurityPolicies>
...
    <ServerSecurityPolicy>
      <SecurityMode>None_1</SecurityMode>
      <SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#None</SecurityPolicyUri>
    </ServerSecurityPolicy>
````

Havlat82, 2020-11-09:
Thanks for ur answer. I have solved this problem with DigiCertUtil app and then restarting computer.

AlinMoldovean, 2020-11-10:
@Havlat82 ,
Please close the issue if it is solved.
Thanks

Havlat82, 2020-11-10:
@AlinMoldovean  I would like to, but I do not see any button / control to close the issue..

closed 2020-11-10 by AlinMoldovean

SyngnathZ, 2022-03-14:
> Thanks for ur answer. I have solved this problem with DigiCertUtil app and then restarting computer.

I got the same problem when using UA.NetStandard in Unity to create a simple server, I've tried to create the cert and private key manually in the %CommonApplicationData%\OPC Foundation\CertificateStores\MachineDefault using "UA Configuration Tool". Can you give us an overview of your solution at that time? I'll appreciate that!