OPCFoundation/UA-.NETStandard #1168: Ambiguous license terms

rustygutter, 2020-10-29:
**Type of Issue**
[ ] Bug [X] Enhancement [ ] Compliance [X] Question [ ] Help wanted

The license terms displayed in the /UA-.NETStandard/Licenses/ folder, and the ones displayed when using NuGet seem to be fairly non-restrictive.  By my reading, I could use the library in a commercial product and not be required to release the source code of my commercial product.  

But then, in the ReadMe ("For more information and license terms, see here."), we are pointed to licensing here: http://opcfoundation.github.io/UA-.NETStandard/.  And there we see pretty clearly that the library has a dual-license -- a non-restrictive one for paying foundation members, and GPLv2 for everyone else.  And my understanding of GPLv2 is that "Any modifications to **or software including (via compiler) GPL-licensed code** must also be made available under the GPL along with build & install instructions.".  

So, which of the contradictory license agreements actually applies to the 'OPCFoundation/UA-.NETStandard' library?

**Expected behavior**
The documentation should be updated to provide unambiguous indication of the licensing that covers its use.

mregen, 2020-10-29:
Hi @rustygutter, your point is correct, the link doesn't fully describe the license terms, they need to be updated on the OPC Foundation web page. The Dual license and MIT applies to the use of the source code in this repository as described in the link, while the redistributable license applies to the nuget packages and assemblies built and signed by the OPC Foundation.

rustygutter, 2020-10-29:
Hi @mregen, thanks for responding.  But at the risk of seeming dense, I don't understand the distinction you're making.  Is the suggestion that someone can use the libraries as distributed w/ NuGet <without> abiding by the dual-license agreement?

To me, that's what section 8 of the license distributed w/ NuGet implies (I'm not lawyer, though).   It states that the NuGet document is  "the entire Agreement for the Software Deliverables and support services".  That would seem to imply no other licensing restrictions (e.g. dual-license) would/could apply.  I was happy to see that as I had heard from others that we (at work) cannot use the OPC Foundation libraries because of the restrictive GPL-v2 terms.  But then in the readme there's that reference to the other licensing page that seems to apply to the same code, only with different license terms.

mregen, 2020-11-14:
Hi @rustygutter, please consult with a lawyer for legal advise on license terms. 

I can only quote that the redistributable license applies to 
> Distributable Code.  The Software Deliverables contain compiled code

While for source in the repo please check this [LICENSE.txt](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/LICENSE.txt)

mregen, 2021-01-15:
closed due to inactivity

closed 2021-01-15 by mregen