OPCFoundation/UA-.NETStandard #1169: PubSup support for UADP

AlinMoldovean, 2020-10-29:
**Type of Issue**
[X] Feature

**Describe the Issue**

Implement a PubSub support for UADP

**Expected behavior**
Provide a library with PubSub implementation for UADP

**Additional context**
With .NET Core support

mpostol, 2020-10-31:
I believe that this work should be reused. Consider harmonization with the:

- #875
- mpostol/OPC-UA-OOI#208
- Azure/Industrial-IoT#38
- TBD

Let me recall you that pubsub has been already implemented and is available as a .NET Standard NuGet package  

- [UAOOI.Networking.SemanticData](https://www.nuget.org/packages/UAOOI.Networking.SemanticData)

```txt
dotnet add package UAOOI.Networking.SemanticData --version 4.0.2
```

This package has been just used to implement: [Azure Gateway Implementation](https://github.com/mpostol/OPC-UA-OOI/tree/master/Networking/DataRepository/AzureGateway#azure-gateway-datarepository-implementation).

I have defined the `NetworkMessage` syntax using Augmented BNF (ABNF). The ABNF syntax definition is [here](https://github.com/mpostol/OPC-UA-OOI/blob/master/Networking/SemanticData/MessageHandling/NetworkMessage.abnf).

Consider joining our effort to promote reusability. Let me know how it works for you.

AlinMoldovean, 2021-01-08:
An initial version of a PubSub library for UDP was added.
We wait for feedback from users.
Further fixes and updated (e.g. after IOP tests) should be added.

closed 2021-01-08 by AlinMoldovean