OPCFoundation/UA-.NETStandard #1174: Opc.Ua.ServiceResultException: BadRequestTimeout at Opc.Ua.SessionClient.Read Random occurances.

AverageWeeaboo, 2020-11-02:
**Type of Issue**
[Bug] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

**Describe the Issue**
I created a read service which attempts to poll once per second but only if there is no other read operation currently occurring (to prevent any form of overload in connections). It usually works fine and measuring the operation, it normally only takes 10-20 ms per cycle read but sometimes it jumps to 200ms and other times it completely fails and times out resulting in the error mentioned in the title. This error did not appear when I used the subscription method to read tags but I require the read method to poll all values at once. I have tested the code both locally with the OPCUA server and remotely but the error still persists across both instances so it is not a network issue as the PLC that the OPC server is reading from is directly connected. Let me know if you need more info

**Code snippets**
If applicable, add a code snippet to help explain your problem.
The Error: 
`Opc.Ua.ServiceResultException: BadRequestTimeout
   at Opc.Ua.Bindings.ChannelAsyncOperation'1.End(Int32 timeout, Boolean throwOnError)
   at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.EndSendRequest(IAsyncResult result)
   at Opc.Ua.SessionClient.Read(RequestHeader requestHeader, Double maxAge, TimestampsToReturn timestampsToReturn, ReadValueIdCollection nodesToRead, DataValueCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at OPCUA_Client.UAClient.ReadNodes() in /mnt/c/Users/toliver.IAI/Documents/VSCode/OPCUA Client/UAClient.cs:line 202
`
Config creation method for client:

> `  public ApplicationConfiguration CreateConfig()

        {
            try{
                return new ApplicationConfiguration()
                {
                  CertificateValidator = new CertificateValidator(),
                    ApplicationName = this.ApplicationName,
                    ApplicationUri = Utils.Format($@"urn:{System.Net.Dns.GetHostName()}:{this.ApplicationName}"),
                    ProductUri = Utils.Format($@"urn:{System.Net.Dns.GetHostName()}:{this.ApplicationName}"),
                    ApplicationType = ApplicationType.Client,
                    SecurityConfiguration = new SecurityConfiguration {
                        ApplicationCertificate = new CertificateIdentifier {
                            StoreType= @"Directory",
                            StorePath= @"Cert/own",
                            SubjectName = Utils.Format($@"CN={this.ApplicationName}, DC={System.Net.Dns.GetHostName()}")
                            },
                        TrustedPeerCertificates = new CertificateTrustList { 
                            StoreType = @"Directory", 
                            StorePath = @"Cert/trusted" 
                            },
                        TrustedIssuerCertificates = new CertificateTrustList { 
                            StoreType = @"Directory", 
                            StorePath = @"Cert/issuer" 
                            },
                        RejectedCertificateStore = new CertificateTrustList { 
                            StoreType = @"Directory", 
                            StorePath = @"Cert/rejected" 
                            },
                        AutoAcceptUntrustedCertificates = false
                    },
                    TransportConfigurations = new TransportConfigurationCollection(),
                    Extensions = new XmlElementCollection(),
                    TransportQuotas = new TransportQuotas {
                        OperationTimeout = this.OperationTimeout,
                        MaxStringLength =1048576,
                        MaxByteStringLength =1048576,
                        MaxArrayLength =65535,
                        MaxMessageSize =4194304,
                        ChannelLifetime = 300000,
                        SecurityTokenLifetime = 3600000
                    },
                    ClientConfiguration = new ClientConfiguration { 
                        DefaultSessionTimeout = this.Timeout,
                        WellKnownDiscoveryUrls = new StringCollection (){
                            $"opc.tcp://{this.ServerAddress}:{this.Port}"
                        },
                        DiscoveryServers = new EndpointDescriptionCollection()     
                    },
                    TraceConfiguration = new TraceConfiguration(){
                        OutputFilePath=@"Logs/Quickstarts.ConsoleReferenceClient.log.txt",
                        DeleteOnLoad =true
                    }
                };
            }
            catch(Exception ex)
            {
                Console.WriteLine(ex.Message);
                return new ApplicationConfiguration();
            }
        } `
**Environment (please complete the following information):**
 - OS: [e.g. Windows 10, Ubuntu, macOS]
**Ubuntu 20**
 - Development environment: [e.g. Visual Studio 2019 16.7.2]
**VSCode**
 - Runtime: [e.g. .NET 4.6.2, .NET Core 3.1]
**.Net Core 3.1.0**
 - OPC Package Release Version [e.g. 1.4.363.107]
 **1.4.363.107**
 - Component: [e.g. Opc.Ua.Core]
**Opc.Ua.Core**
 - Server: [e.g. Reference Server]
**Ignition Server.**
 - Client: [e.g. UA Expert]
**Custom Client based on the reference console client**

AlinMoldovean, 2020-11-04:
Hello @Thomas-Oliv ,

The BadRequestTimeout is generated due to unresponsive server.
The client side seems ok, but the server does not respond to Read service request within the specified timeout.
This is something that needs to be investigated on the server side, i.e. **Ignition Server**.

As a test, you can try to make a cyclic read on Reference Server and check if the issue occurs.

closed 2021-04-05 by AlinMoldovean

guerreromarvin, 2025-02-19:
Hi,

  I am using UacPP serve running on the same machine. There shouldn't be any problem with the server. For my case, the exception only happens when data is not received from the server, which is normal for slow moving values.  Even if the server does not send data and the client think it should, the client must never throw an exception that would cause threads to pause even by milliseconds.  In my case everything stops for about 3 seconds before resuming normal operation. This kind of exception must be quite and in my opinion should just raise an event and not an exception.