OPCFoundation/UA-.NETStandard #1176: Subscription Reconnect

xky0007, 2020-11-06:
**Type of Issue**
[Subscription Reconnection ] Help wanted

**Describe the Issue**

I have a client that has several subscriptions. Each subscription may contain some monitored items.
What is the correct way to restore all those subscriptions with correct monitored items once the client session disconnected from the server for some reason. For example, the server reboot.

**Expected behavior**
All subscriptions and monitored items work smoothly.

MD-V, 2020-11-06:
If subscription.CurrentKeepAliveCount < subscription.KeepAliveCount the Subscription should start updating on their own as soon as the reconnect happpens. 

if subscription.CurrentKeepAliveCount >= subscription.KeepAliveCount you will have to setup and create the subscriptions again.

Zigby, 2020-11-06:
See the implementation in SessionReconnectHandler.cs in OPC.Ua.Client. The Reconnect and Recreate methods of the Session class handles the subscriptions. In case of a recreate (e.g. due to server reboot) the subscriptions needs to be recreates as well, but this is handled inside the Recreate method of the Session class.

AlinMoldovean, 2020-11-06:
@xky0007 ,

An example of using ReconnectHandler can be found here: https://github.com/OPCFoundation/UA-.NETStandard/issues/999#issuecomment-670520049

xky0007, 2020-11-06:
> See the implementation in SessionReconnectHandler.cs in OPC.Ua.Client. The Reconnect and Recreate methods of the Session class handles the subscriptions. In case of a recreate (e.g. due to server [...]

Found in `public static Session Recreate(Session template)`. Thanks

closed 2020-11-06 by xky0007