OPCFoundation/UA-.NETStandard #1177: Not possible to handle multiple sessions on a single channel

Zigby, 2020-11-06:
**Type of Issue**
Enhancement (or possibly a bug?)

**Describe the Issue**
Not possible to handle multiple sessions on a single channel using the sdk as client (server side works fine). It is not possible to dispose a Session without disposing the channel.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a SecureChannel (SessionChannel.Create()
2. Create Session1 passing the channel in the ctor.
3. Create Session2 passing the channel in the ctor
3. Dispose Session1.
4. Calling Dispose on Session1 disposes the channel and hence Session2 no longer has a channel

**Expected behavior**
Expected behavior would be that I can control the lifetime of the channel rather than the session taking ownership of the channel, e.g. by calling a method DetachChannel prior calling Dispose the Session.
Below is a code snippet of the additions I did to the Opc.Ua.Core to make this work (added an AttachChannel and Detatch channel method to the ClientBase class).

**Code snippets**
A solution to the problem would be to add methods to attach/detach a channel to a session.
```
class ClientBase
{
…
    public void DetachChannel()
    {
        m_channel = null;
    }

    public void AttachChannel(ITransportChannel channel)
    {
        m_channel = channel;
    }
...
}
```

I have tested adding these methods and with these additions it works fine to Open/Close/Reconnect/Recreate sessions using a single channel (tcp connection) to the server. This solution preserves the behavior that the Session owns the channel, but allows a client (caller) to explicitly detach the channel from the Session.

**Environment (please complete the following information):**
 - OS: Windows 10 
 - Development environment: Visual Studio 2019 16.7.2
 - Runtime: .NET Core 3.1
 - OPC Package Release Version: 1.4.363.107
 - Component: Opc.Ua.Core
 - Server: Any Server (tested with a server built with this SDK).
 - Client: My own client using components from this repo..

**Additional context**
The Scenario for this is a service that handle many client connections (web clients) and it needs to sessions towards downstream OPC UA server with the identity of the client. For this scenario it is desired to keep a single connection to downstream server, and just create sessions based on user identity.

mrsuciu, 2021-09-14:
Can you contribute by providing a Pull Request with your solution ?

Zigby, 2021-09-17:
Yes, I will start working on it next week. Thanks for taking this into consideration.

mregen, 2022-02-09:
Just fyi, yesterday there was such a topic discussed in the compliance group, the thinking is it is allowed to have multiple session in a secure channel, however some server may not support it or have different requirements or even have only limited number of secure channels. I *think* that makes a case to support your enhancement, just the client needs to deal with the non mandatory support. Once we release 368 it can be picked up again.

closed 2022-04-01 by mregen