OPCFoundation/UA-.NETStandard #1180: Create a client with a Session includes some Subscriptions

xky0007, 2020-11-09:
**Type of Issue**
[ ] Help wanted

**Describe the Issue**

I'm following the example to create a client that contains some subscriptions. I met the situation that the server may not be available at the moment I create the Client. 

Because the example used `var selectedEndpoint = CoreClientUtils.SelectEndpoint(endpointURL, false, 15000);` which require the server is on before the session creation. I can't create a client with a disconnected session - which will automatically reconnect and recreate all related subscriptions whenever the OpcUa server is ready. [Subscription Reconnect  #1176](https://github.com/OPCFoundation/UA-.NETStandard/issues/1176).

What is the correct way to create a disconnected session client?

Thanks.

**Code snippets**

```
ApplicationConfiguration config = BuildAppConfig();
var selectedEndpoint = CoreClientUtils.SelectEndpoint(endpointURL, false, 15000);
config.CertificateValidator.CertificateValidation += CertificateValidator_CertificateValidation;
var endpointConfiguration = EndpointConfiguration.Create(config);
var endpoint = new ConfiguredEndpoint(null, selectedEndpoint, endpointConfiguration);
session = await Session.Create(config, endpoint, false, $"session-{appName}", 60000, new UserIdentity(new AnonymousIdentityToken()), null);
session.KeepAlive += Client_KeepAlive;
```

**Expected behavior**
A client contains a disconnected session and it will reconnect once the server is ready.

AlinMoldovean, 2020-11-09:
Hello @xky0007 ,

`Session.Create()` fails in case the server is not available at that moment.
The best way is to implement a waiting loop until the server is available.

xky0007, 2020-11-10:
> Hello @xky0007 ,
> 
> `Session.Create()` fails in case the server is not available at that moment.
> The best way is to implement a waiting loop until the server is available.

The code won't execute `Session.Create()` because it failed at `CoreClientUtils.SelectEndpoint()`. Is there any example to manually set `EndPointDescription` and `ConfiguredEndPoint`? So I can use `new Session()` to created a session with disconnected channel.

Thanks:D

AlinMoldovean, 2020-11-10:
@xky0007 ,

You can have a `ConfiguredEndpoint` serialized in a xml file.
An example on Samples repo [here ](https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/master/Samples/Client.Net4/Opc.Ua.SampleClient.Endpoints.xml):

closed 2021-02-04 by AlinMoldovean