OPCFoundation/UA-.NETStandard #1186: Need Help with Certificate Creation for Application.

MAF63, 2020-11-13:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [ X] Question [ X] Help wanted
  

I inherited an application that was developed based on code from the OPCUA Foundation SDK.
I'll admit upfront I know very little about certificates in OPCUA.

The application is used to communicate with PLCs via Ignition OPCUA server.
The app was developed last year and is still working on a test machine.
However, in the dev environment, the app will no longer work in visual studio
I am now getting certificate related error messages.

I originally was receiving a "Certificate has is expired or not yet valid" message.
I found where the certs are located ( C:\Users\...\AppData\Local\OPC Foundation\pki\trusted\certs\)
and indeed found a cert that expired last week.

I did some reading and attempted to reissue the cert with UA Configuration tool.
It did create a certificate ( and two additional seperate directories) and the certificate does not expire for 10 years.

However when I compile and run the app I get the following : 

<11/13/2020 2:07:17 PM> System.AggregateException: One or more errors occurred. ---> Opc.Ua.ServiceResultException: BadSecurityChecksFailed ---> Opc.Ua.ServiceResultException: BadSecurityChecksFailed
   at Opc.Ua.CertificateValidator.<InternalValidate>d__22.MoveNext()
   --- End of inner exception stack trace ---
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain)
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2 certificate)
   at Opc.Ua.Configuration.ApplicationInstance.<CheckApplicationInstanceCertificate>d__53.MoveNext()

mregen, 2020-11-14:
Hi @MAF63, if you delete the cert and private key in pki/own your app will automatically create a new certificate on the next start. But also the certificate on the other device may have to be renewed.
After renewing certificates trust needs to be reestablished by moving the corresponing cert from a 'rejected' to a 'trusted' folder after a rejected connection attempt. Hope this helps.

MAF63, 2020-11-16:
Thank you for response.
I did delete  all certificates and the private key in the directory  C:\Users...\AppData\Local\OPC Foundation\pki\ 
I restarted the code in  Visual Studio and unfortunately,  no new certificates were created. 

Still getting : _System.AggregateException: One or more errors occurred. ---> Opc.Ua.ServiceResultException: BadSecurityChecksFailed ---> Opc.Ua.ServiceResultException: BadSecurityChecksFailed_

Not sure how to proceed at this point.

MAF63, 2020-11-16:
Update : **Success !**
The certificate is now being created by the application ! 

I'm not entirely sure what I specifically did to allow this to happen, After deleting all certs and private keys etc and trying to create a key by running the application, I  reread the page at  https://github.com/OPCFoundation/UA-.NETStandard-Samples

I read where certificates for UWP apps are stored in **CurrentUser\UA_MachineDefault.**  I opened  the Windows Certificate Management Console and searched of "OP Core" and found my old certs. I then deleted them all.

Now after deleting the certificates it seems that  my application readily generates  a certificate.  I can delete the certificate from
\AppData\Local\OPC Foundation\pki\trusted\certs  and then start the app and a new one is generated.

Furthermore, the certificate will show up in the Ignition OPCUA quarantined server certificate area, it can be trusted and the application works as it expected to.

My question is should the deleting of the keys in **CurrentUser\UA_MachineDefault.**  have an impact of my app being able to create a cert ?

mregen, 2020-11-16:
Hi @MAF63, it depends on the configuration. The app certs can be stored in the X509Store or in the DirectoryStore. For X509Store they end up where you found them. If there is still an expired certificate in the store, it is not replaced, because it could still be used. Expired certificates are a suppressable error. So manual operation to delete them is necessary. The use case is not well handled in our samples, I believe. There should be a dialog with the user asking to renew the cert.
The copy of the store in trusted certs is just copied for convenience, if other OPC apps on the same PC should trust each other by default. It can be suppressed as well by an option... Cheers!

MAF63, 2020-11-17:
Thanks again for the assistance - it was appreciated !

closed 2020-11-17 by MAF63