OPCFoundation/UA-.NETStandard #1188: SessionlessServiceMessage class is not implemented according to spec

mregen, 2020-11-18:
**Type of Issue**
[X] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

The `SessionLessServiceMessage` misses the urisVersion field

https://reference.opcfoundation.org/v104/Core/docs/Part4/6.3.2/

The urisVersion field is a VersionTime.

The urisVersion is misspelled in the latest Nodeset as an array. To be fixed in 1.04.8

mregen, 2021-02-24:
waiting for 1.4.9 errata

AlinMoldovean, 2021-04-02:
@mregen,
A fix was added in #1284 
Is it enough to close this?

```
        /// <remarks />
        [DataMember(Name = "UrisVersion", IsRequired = false, Order = 1)]
        public uint UrisVersion
        {
            get { return m_urisVersion;  }
            set { m_urisVersion = value; }
        }
```

mregen, 2021-04-06:
I *think* BuiltIn/SessionlessServiceMessage.cs and the test are still not quite right...

mregen, 2021-09-04:
with latest Errata it should be fixed..

PR #1505 (2021-09-13): Add missing vars in session less message
fixes #1188
closed 2021-09-13 by mregen