OPCFoundation/UA-.NETStandard #1189: Connect with SINUMERIK828D return "BadIndentityTokenRejected"

SmokeCircle, 2020-11-19:
**Type of Issue**
 [ ] Help wanted

**Describe the Issue**
I want to connect with SINUMERIK 828D OPCUA:
I use the "NetCoreConsoleClient" to connect:
The console result is:

.Net Core OPC UA Console Client sample
1 - Create an Application Configuration.
2 - Discover endpoints of opc.tcp://192.168.1.10:4840.
    Selected endpoint uses: Basic128Rsa15
3 - Create a session with OPC UA server.
Accepted Certificate: CN=ncu1, E=opcua@siemens.com, OU=Industry, O=Siemens AG, L=Erlangen, S=Bavaria, C=DE
Exception: One or more errors occurred. (BadIdentityTokenRejected)

My session code is:
session = await Session.Create(config, endpoint, false, "OPC UA Console Client", 60000, new UserIdentity("username","password"), null);

But it is OK to connect with "Kepware" OPC UA server.

Do I miss something?

AlinMoldovean, 2020-11-19:
Hi @SmokeCircle ,

The `BadIdentityTokenRejected ` error returned by server suggests that the provided user identity for the session is not accepted by the UA Server (i.e. user: "_username_" and password: "_password_")
You need to get from the server side the valid user credentials.

SmokeCircle, 2020-11-20:
Thanks @AlinMoldovean 
I am sure my "username" and "password" is correct.
And I found the exception is thorwn in the method :
"private void ValidateServerNonce(IUserIdentity identity, byte[] serverNonce, string securityPolicyUri, byte[] previousServerNonce)"
It is in Opc.Ua.Client  "session" class.
I don't understand the meaning of the input parameters "byte[] serverNonce" and "byte[] previousServerNonce".
In my project, the "byte[] serverNonce" always returns null,and  the "byte[] previousServerNonce" is always a byte[1] array, and the value is 1.
I just ignore the "ValidateServerNonce" function and everything is OK.
But I don't understand why?

AlinMoldovean, 2020-11-20:
@SmokeCircle ,

`ValidateServerNonce` methods checks if the nonce returned by the server is valid.
In you case you try to connect to a UA Server that does not comply with the OPC UA specification for nonce handling.
And `ValidateServerNonce()` will throw a `BadNonceInvalid` error and not `BadIdentityTokenRejected`.

closed 2021-03-16 by AlinMoldovean