OPCFoundation/UA-.NETStandard #1191: ServiceMessageContext instance used at NodeStateCollection loading / saving methods is wrong

Hevgen, 2020-11-19:
**Type of Issue**
Bug

We use _NodeStateCollection_ instance and its methods _LoadFromBinary(), LoadFromXml(), SaveAsBinary(), SaveAsXml()_ to load/save some nodes from/to files.
The actual implementation of these methods doesn't have arguments of type _ServiceMessageContext_ and creates ServiceMessageContext instances in the wrong way inside the methods: it uses default settings instead of actually used at the server.

Sample of actual codes:

```
public void SaveAsBinary(ISystemContext context, Stream ostrm)
        {
            ServiceMessageContext messageContext = new ServiceMessageContext();

            messageContext.NamespaceUris = context.NamespaceUris;
            messageContext.ServerUris = context.ServerUris;
            messageContext.Factory = context.EncodeableFactory;

            BinaryEncoder encoder = new BinaryEncoder(ostrm, messageContext);
```

So next ServiceMessageContext properties just got default values:
```
      MaxStringLength =UInt16.MaxValue;
      MaxByteStringLength = UInt16.MaxValue*16;
      MaxArrayLength = UInt16.MaxValue;
      MaxMessageSize = UInt16.MaxValue*32;
      MaxEncodingNestingLevels = 200U;

```

The quick-fix proposition is to add argument _ServiceMessageContext messageContext = null_ into the methods. That will allow posting correct context from outside and doesn't break existing codes.

AlinMoldovean, 2021-01-15:
Hi @Hevgen ,

Could you provide a PR for this fix?

Hevgen, 2021-01-15:
I will try to do PR during the next few weeks.

AlinMoldovean, 2021-03-31:
@Hevgen ,
Please reopen if you provide a PR for this

closed 2021-03-31 by AlinMoldovean