OPCFoundation/UA-.NETStandard #1193: ManagedApplication class not found in Opc.Ua.Configuration in new version Version=1.4.362.42

PrashantDesai, 2020-11-25:
We used ManagedApplication class for Certificate Management export / import

- in Assembly Opc.Ua.Configuration, Version=1.4.358.0, ..\.nuget\packages\opcfoundation.netstandard.opc.ua\1.4.358.30\lib\net46\Opc.Ua.Configuration.dll

- After upgrading to Version=1.4.362.42 same class not found

Please provide alternative class

MD-V, 2020-11-25:
Class was removed in https://github.com/OPCFoundation/UA-.NETStandard/pull/985 because it was not needed. If you need it just copy the code from the last version that it was included in.

mregen, 2021-01-12:
please reopen if there are still questions... if there is a lot of demand we can bring back the class, but imho it was unused code.

closed 2021-01-12 by mregen

PrashantDesai, 2021-01-13:
It is ok, we have added the file ManagedApplication.cs. But any license issue ? since using direct code.
Otherwise close it.

mregen, 2021-01-13:
I believe the source for managedapplication is under MIT license, so no license issue.

PrashantDesai, 2021-01-13:
Yes, Thank you