OPCFoundation/UA-.NETStandard #1196: BuiltInType and ValueRank are wrong on a variable with type byte[]

marcelwinh, 2020-11-30:
**Type of Issue**
[X] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

**Describe the Issue**

When you read a value with the client it looks like the builtInType and ValueRank are interpreted from the value you receive.
If i read the value from a byte array i receive BuiltInType = 15 (ByteString) and ValueRank = -1.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Sample 'Opc.Ua.SampleClient (Universal Windows)' https://github.com/OPCFoundation/UA-.NETStandard-Samples
	add the following lines e.g. to the ClientPage.xaml.cs in the onClick event of the Report Button line 242.
		var value = m_session.ReadValue((NodeId)reference.NodeId);
		var node = m_session.ReadNode((NodeId)reference.NodeId);
2. Connect to Server 
3. Read value of a Node with data type byte array
4. See error
received value is a byte[] but BuiltInType(15) & ValueRank(-1) are invalid

**Expected behavior**
I would expect that a one dimensional byte array has the BuiltIntType = 3 and ValueRank = 1

**Screenshots**
Array of Bytes in Wireshark
[image]

Array of bytes seen in C# app
[image]

ByteString in WireShark
[image]

Array of ByteString in Wireshark
[image]

**Code snippets**
add the following lines e.g. to the ClientPage.xaml.cs in the onClick event of the Report Button line 242.
```C#
		var value = m_session.ReadValue((NodeId)reference.NodeId);
		var node = m_session.ReadNode((NodeId)reference.NodeId);
```
**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2017 15.9.27
 - Runtime: .NET 4.6.2
 - OPC Package Release Version 1.4.363.107 & 1.4.362.42
 - Component: DataValue IOpcUaSession.ReadValue(NodeId nodeId)
 - Server: netFIELD IO-Link OPC UA Server and a node-opcua based server
 - Client: Opc.Ua.SampleClient (Universal Windows)

AlinMoldovean, 2020-12-03:
Hello @marcelwinh ,

This stack treats one-dimensional Byte array as ByteString. 
This statement from the OPC UA Specs describes the case:

   **5.1.4 ByteString
A ByteString is structurally the same as a one-dimensional array of Byte. It is represented as a distinct built-in data type because it allows encoders to optimize the transmission of the value. However, some DevelopmentPlatforms will not be able to preserve the distinction between a ByteString and a one-dimensional array of Byte.
If a decoder for DevelopmentPlatform cannot preserve the distinction it shall convert all one-dimensional arrays of Byte to ByteStrings.**

mregen, 2021-01-12:
please reopen if there are still questions...

closed 2021-01-12 by mregen