OPCFoundation/UA-.NETStandard #1198: OPC Sessions not closing causing BadTooManySessions error

lkomanetz, 2020-12-01:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [ ] Question [x] Help wanted

**Describe the Issue**

After a period of time the client application fails to connect to OPC Server with "BadTooManySessions" exception.  This happens even though `OpcSession.Close()` is called when the client application closes.

**To Reproduce**
Steps to reproduce the behavior:

1. Have a client application connect and establish an OpcSession
2. Ensure `OpcSession.Close()` is called when the client application closes
3. Close and reopen multiple times until `BadToomanySessions` occurs.

**Expected behavior**
The established session should close and should not linger in the server.

**Screenshots**
N/A

**Log output**
No log messages are provided when this happens.

**Code snippets**
*Opening a session*
```cs
        public async Task OpenAsync() {
            var opcUrl = $"opc.tcp://{IpAddress}:{Port}";
            var appConfig = await LoadOpcConfigurationAsync().ConfigureAwait(false);
            var endpoint = ConfigureEndpoint(opcUrl, appConfig);
            _session = await Session.Create(
                configuration: appConfig,
                endpoint: endpoint,
                updateBeforeConnect: false,
                sessionName: GetSessionName(),
                sessionTimeout: 60_000,
                identity: null,
                preferredLocales: null
            );

            static ConfiguredEndpoint ConfigureEndpoint(string url, ApplicationConfiguration appConfig) {
                var selectedEndpoint = OpcClientUtils.SelectEndpoint(url, false, 15_000);
                var endpointConfig = EndpointConfiguration.Create(appConfig);
                var endpoint = new ConfiguredEndpoint(selectedEndpoint.Server, endpointConfig);
                endpoint.Update(selectedEndpoint);
                return endpoint;
            }

            static string GetSessionName() {
                var fullyQualifiedName = typeof(OpcClientSession).FullName;
                return fullyQualifiedName.Substring(0, fullyQualifiedName.IndexOf('.'));
            }
    }
```
*Closing the session*
```cs
        public void Close() {
            var statusCode = _session.Close();
            if (StatusCode.IsNotGood(statusCode)) throw new Exception("Unable to disconnect from OPC session.");
        }
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.8
 - Runtime: .NET 5
 - OPC Package Release Version 1.4.361.39
 - Component: [e.g. Opc.Ua.Core]
 - Server: Emerson PLC (built-in server)
 - Client: Custom

**Additional context**
N/A

AlinMoldovean, 2020-12-03:
Hello @lkomanetz ,

__session.Close()_ call will invoke a _CloseSession()_ call on the server ion case the session is connected.
The reason why the server does not release the sessions should be investigated on the _Emerson PLC UA Server_ side.
The issue does not replicate when testing with Reference Server from this repository.

lkomanetz, 2020-12-03:
Awesome.  Thank you!

closed 2020-12-03 by lkomanetz