OPCFoundation/UA-.NETStandard #1199: Need information regarding configuring certificates in an application

MAF63, 2020-12-01:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [X] Question [X] Help wanted

I am looking for information  on how to set certificate configuration in an application.
I have done some google searches but have not found a "how to" guide.
Does anyone have a link to a guide that can answer questions like 
- 	Set how long a certificate is valid for
- 	Where will a certificate be stored - cert placement in X509Store or in the DirectoryStore. 
- 	How to set name and other information on certificate to reflect issuing entity 
- 	How to suppress certain errors etc.

Thanks !

mregen, 2020-12-22:
Hi @MAF63, to answer your questions:
- currently I would recommend a maximum 2 year period for 2k RSA application certificates 
- either store has advantages, X509Store for app certs is more secure especially on windows its the windows cert store. For trustlist directory store is recommended (X509Store doesn't support CRLs and is harder to manage)
- issueing entities cannot be set, just the subject DN. For issuer signed app certificates the issuer cert must be available in the issuer or trusted store. The X500 DN subject name can be set in the configuration file  --> SubjectName
- Errors can be suppressed by implementing the callback for teh cert validator, by setting configuration flags (e.g. to suppress Sha 1 errors) or by setting ValidationOptions with the issuer/trusted store configuration.

mregen, 2021-01-12:
please reopen if there are still questions...

closed 2021-01-12 by mregen