OPCFoundation/UA-.NETStandard #1200: Application runs fine stand alone but generates error when run as a service

MAF63, 2020-12-01:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [ ] Question [X] Help wanted

Has anyone seen this before ?
I can run a C# console application using OPCUA on its own  fine. 
But when run as service I get error : "Certificate has expired or is not yet valid."

Steps to reproduce the behavior:
1. Service is configured to run as administrator 
2. Start service.
3. Error is logged

**Expected behavior**
Service should start without errors.

**Log output**
When run as application there is no error generated..

When run as service the application error log contains : 
<12/1/2020 9:32:00 AM> System.AggregateException: One or more errors occurred. ---> Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: A required certificate is not within its validity period when verifying against the current system clock or the timestamp in the signed file.

**Environment 
 - Windows 10
 - Visual Studio 2019 
 - .NET 4.6.2, .NET Core 3.1]
 - Component: [e.g. Opc.Ua.Core]
 - Server:  Ignition Version: 8.0.4 (b2019091612)
- Client C# application

Hevgen, 2020-12-01:
Seems it is not service-specific.
It is about a certificate. Try to delete existing one (look at folder according to your OPC-UA Application config). On next application start fresh certificate will be generated.
At least I got such error on console application and I have Service working ok now.
BTW. What type of application do you have: server or client? Do you get error on Application.CheckApplicationInstanceCertificate()?

JeremyMahieu, 2020-12-07:
Typically services run as a different user as you said, it runs as Administrator. This user may not have access to the same certificate your user account has access to. Have you installed certificates under your Windows User certificate store that you didn't install for the Computer account certificate store?

IIn start enter "mmc" and add the Certificates snap-in, you can choose several `accounts`. Try putting the relevant certificates in the `Computer account` so every user can use them.

[image]

MAF63, 2020-12-09:
The application is a client. No, the error is not thrown during Application.CheckApplicationInstanceCertificate().

Also, when  I delete the certificate, I restart app and find new cert. I must also then trust the quarantined cert in Ignition server.
I then restart my app and things run,  I try the same by starting app as service and get same error as always.

Hevgen, 2020-12-09:
> Also, when I delete the certificate, I restart app and find new cert. I must also then trust the quarantined cert in Ignition server.

So you deleted certificate of YOUR application and trust new one at the server - the problem is not about it.
You have client application and got Opc.Ua.ServiceResultException - looks like it is about communication establishing. So what about the certificate of "Ignition server" got by your app? Have you looked at it? (In my cases I find bad server certificates into RejectedCertificates\certs folder according to my client config file).

closed 2021-03-31 by AlinMoldovean