OPCFoundation/UA-.NETStandard #1201: CTT tests fail on ReferenceServer, ConsoleReferenceServer

kimaluco17, 2020-12-01:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [x] Question [ ] Help wanted

**Describe the Issue**

The Reference server fails the following CTT Attribute Services tests: Attribute Alternate Encoding\Err-004.js, Attribute Alternate Encoding\Err-005.js, Attribute Write Values\003.js

**To Reproduce**
Steps to reproduce the behavior:
1. Launch either reference servers (.NET Core or Framework)
2. Launch the CTT with the UAReferenceServer.ctt.xml configuration
3. Select the following tests: Attribute Alternate Encoding\Err-004.js, Attribute Alternate Encoding\Err-005.js, Attribute Write Values\003.js
4. The above tests fail as shown in screenshots below.

**Expected behavior**
If the reference server passed compliance, I would expect those tests to pass. There may be configuration issues on my side that I am not aware of or those CTT tests are badly formed.

**Screenshots**
[image]
[image]
[image]

**Log output**
Err-004.js: 
Response.Results[0].StatusCode = Good (0x00000000) but 'BadDataEncodingUnsupported (0x80390000)' was expected

Err-005.js: 
Response.Results[0].StatusCode = Good (0x00000000) but 'BadDataEncodingUnsupported (0x80390000)' was expected

003.js:
Testing type: ByteString
Read.Response.Results (count: 5) with RequestHandle=14 StatusCode check (suppressed):
	[0]  (Setting: '/Server Test/NodeIds/Static/All Profiles/Scalar/ByteString') StatusCode is Good (0x00000000),
	...,
	[4]  (Setting: '/Server Test/NodeIds/Static/All Profiles/Scalar/ByteString') StatusCode is Good (0x00000000)
Expected a value that was previously written. According to the specification a write should return after the write operation 
has been completed otherwise the server needs to return a Good_CompletesAsynchronously.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: N/A
 - Runtime: .NET 4.6.2, .NET Core 3.1
 - OPC Package Release Version: 1.4.364.31
 - Component: N/A
 - Server: ConsoleReferenceServer, ReferenceServer(Quickstarts)
 - Client: CTT v1.4.9.396

AlinMoldovean, 2020-12-08:
Hello @kimaluco17 ,

The "_Attribute Alternate Encoding_" tests are not included in the profiles used for compliance testing, i.e. "_Standard 2017 UA Server_", "_Data Access Server Facet_" and "_Method Server Facet_".
We need to check this.

The test "Attribute Write Values\003. js" it's a known issue reported also in  #1086 .
We intend to fix this, but it requires a breaking change and we need to handle that properly.

kimaluco17, 2021-01-06:
@AlinMoldovean 

Thank you for the reply, I had realized shortly after posting this that the "Attribute Alternate Encoding" is in fact not part of the Standard 2017 UA Server profile like you stated.

As for the "Attribute Write Values\003.js" issue, I also found a Mantis issue on the CTT relating to this so I am assuming that is okay to fail for now: http://apps.opcfoundation.org/mantis/view.php?id=4376

closed 2021-01-06 by kimaluco17