OPCFoundation/UA-.NETStandard #1202: OPC UA Security and HSM hardware

eoursel, 2020-12-02:
**Type of Issue**
[ ] Bug [ X] Enhancement [ ] Compliance [X ] Question [X ] Help wanted

Just want to know if someone has already tried to store the certificates in a HSM hardware. I know this is not supported today by the security layer of  the OPC UA stack, but do you have an idea of the work to be done to support this kind of storage of private keys? Especially is it possible to extend the stack without breaking it but introducing a new Certificate store different from DirectoryCertificateStore and X509CertificateStore? We have to write also a new CertificateFactory also ? 

@randy-armstrong  this is a question for your talk _deep dive about OPC UA security_ ;-)

@mregen @barnstee,  what is the interface between  your GDS and Azure Key Vault? Is it something generic that could be reused and adapted with another HSM hardware?  

Contact me privately for some more detailed information about my use case. Thanks in advance.

barnstee, 2020-12-04:
Yes, I built a PoC for this using a TPM, see here: https://github.com/barnstee/UA-.NETStandardLibrary/blob/master/Stack/Opc.Ua.Core/Security/Certificates/TPMCertificateStore.cs

mregen, 2021-01-12:
Hopefully this question is answered... at this time no plan to add PKCS#11 support.

closed 2021-01-12 by mregen