OPCFoundation/UA-.NETStandard #1207: Randomly losing connection with device using keepalive

JeremyMahieu, 2020-12-07:
**Type of Issue**
[x] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

**Describe the Issue**

This is an application in a production environment. It's a Siemens RF680R RFID reader which is OPC UA Server. We connect and periodically (200ms) read one tag.

Serveral times a day we get an exception after which the code reconnects and the process continues. Don't know if this exception is cause or effect of the problem.
`Exception thrown: 'System.ObjectDisposedException' in Opc.Ua.Core.dll`
After this `SessionClient.Connected` is false.

We do use a keepalive, and the `KeepAliveEventArgs` must not have been `ServerState.Running`. If it's not running I do 
```
            if ( _session != null )
            {
                try
                {
                    _session.KeepAlive -= KeepAliveEventHandler;
                    if (_sessionH != null) { _sessionH.Dispose(); }
                    _session.Dispose();
                }
                catch
                {

                }
            }
```

What would be the cause of the keepalive event args not being `Connected` and what is the correct way of closing a failed sessions instead of just `Dispose`.

Might get more info later, I'll update the issue if I do.

**To Reproduce**
Cannot reproduce it on demand but it happens daily so there's opportunity to further debug. Any idea what I could do?

**Expected behavior**
Device should not disconnect.

**Log output**
`Exception thrown: 'System.ObjectDisposedException' in Opc.Ua.Core.dll`

**Environment (please complete the following information):**
 - OS: Windows Server 2019 Standard
 - Development environment: Visual Studio 2019 Version 16.8.2
 - Runtime: .NET Core 3.1
 - OPC Package Release Version: 1.4.363.107
 - Component: Opc.Ua.Core
 - Server: Siemens RF680R RFID Reader
 - Client: Own code based on reference client

AlinMoldovean, 2020-12-08:
Hello @JeremyMahieu ,

`BadSessionIdInvalid ` should mean that the session was disposed by the server.
`BadSessionNotActivated ` should be returned when the reconnect procedure attempts to check if the session still exists on server side by invoking an `ActivateSession` service call.

JeremyMahieu, 2020-12-08:
I found what the cause of this was. Server timeout was set to 6000ms, while default keepalive is 5000ms. Now and then my application would hang and cause the keepalive to come too late (6500ms for example). The very next keepalive gives `BadSessionNotActivated` the one after that `BadSessionIdInvalid`. 

Increasing the timeout to 20s fixed the disconnects. This way there is enough margin between the keepalive and a timeout.

For me this is fixed, don't know if you want to keep it open to figure out why the not activated error? Sounds like a Siemens problem instead of a OPC UA problem at this point.

AlinMoldovean, 2020-12-08:
@JeremyMahieu ,

It is normal to see some errors during the reconnect procedure of this client in case the session was disposed on the server side.
That is because the _ActivateSession_ used to check if the session exist on server side will fail.

I will close this issue.

closed 2020-12-08 by AlinMoldovean

ChristianGrabmann, 2024-10-01:
Hi, we faced the same issue after updating from 1.4.371.96 to 1.5.374.36
After some time we had the same problem as [JeremyMahieu](https://github.com/JeremyMahieu) (see top)
With 1.4.371.96 we had no problems.

Does anyone have any ideas, if there were changes in the software which could cause this behaviour?