OPCFoundation/UA-.NETStandard #1215: Finding servers on network without Discovery server

manfred-evon, 2020-12-11:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [x] Question [ ] Help wanted

**Describe the Issue**

Hello, is there a way to find OPC UA servers on a network without a Discovery server using the UA .NETStandard stack?

AlinMoldovean, 2020-12-14:
Hello @manfred-evon,

In order to find registered OPC UA server in a network you need to use a UA Discovery server with multicast extensions, i.e. [LDS-ME](https://opcfoundation.github.io/UA-LDS/) 

Without UA Discovery Server you need to know the machine where the UA server is running and the port number in order to call a GetEndpoints()

manfred-evon, 2020-12-14:
@AlinMoldovean Thank you very much for your answer. Is it possible to find these UA Discovery servers without specifying a Discovery URL?
I ask this because I want to find all OPC UA servers on a network which are reachable from a specific client and I don't want to let the user enter a Discovery URL.

AlinMoldovean, 2020-12-14:
@manfred-evon ,

A UA Discovery server installed on a specific machine will run on port 4840.
You need to replace "localhost" from the code bellow with an IP address or machine name in order to check if a UA Discovery is running on that host, and get Urls of the registered UA Server instances.

```
        public static IList<string> DiscoverServers(
            ApplicationConfiguration configuration,
            int discoverTimeout
            )
        {
            List<string> serverUrls = new List<string>();

            // set a short timeout because this is happening in the drop down event.
            var endpointConfiguration = EndpointConfiguration.Create(configuration);
            endpointConfiguration.OperationTimeout = discoverTimeout;

            // Connect to the local discovery server and find the available servers.
            using (DiscoveryClient client = DiscoveryClient.Create(new Uri(String.Format(Utils.DiscoveryUrls[0], "localhost")), endpointConfiguration))
            {
                ApplicationDescriptionCollection servers = client.FindServers(null);

                // populate the drop down list with the discovery URLs for the available servers.
                for (int ii = 0; ii < servers.Count; ii++)
                {
                    if (servers[ii].ApplicationType == ApplicationType.DiscoveryServer)
                    {
                        continue;
                    }

                    for (int jj = 0; jj < servers[ii].DiscoveryUrls.Count; jj++)
                    {
                        string discoveryUrl = servers[ii].DiscoveryUrls[jj];

                        // Many servers will use the '/discovery' suffix for the discovery endpoint.
                        // The URL without this prefix should be the base URL for the server. 
                        if (discoveryUrl.EndsWith("/discovery"))
                        {
                            discoveryUrl = discoveryUrl.Substring(0, discoveryUrl.Length - "/discovery".Length);
                        }

                        // ensure duplicates do not get added.
                        if (!serverUrls.Contains(discoveryUrl))
                        {
                            serverUrls.Add(discoveryUrl);
                        }
                    }
                }
            }

            return serverUrls;
        }
```

mregen, 2021-01-12:
please reopen if there are still questions

closed 2021-01-12 by mregen