OPCFoundation/UA-.NETStandard #1218: CTT manual test cases

GregoireG-C, 2020-12-16:
**Type of Issue**
[ ] Bug [ ] Enhancement [X] Compliance [X] Question [ ] Help wanted

**Describe the Issue**

I'me wondering what you are expecting for the test cases noticed as "manual" in the CTT.
I am working to make a software certified via the Standard 2017 UA server profile and some of the CTT profiles contains test cases not implemented with a message: "Please test this CU manually".

What are you expecting us in the objectives to gain the certification ?

Environment:
 - OS: Windows 10
 - Development environment: Visual Studio 2019 16.7.2
 - Runtime: NET 4.6.2
 - OPC Package Release Version 1.4.363.42
 - CTT version: 1.4.04

AlinMoldovean, 2020-12-16:
Hi @GregoireG-C,

A full description of the CTT test can be found on a PC where CTT is installed at this location:
file:///C:/Program%20Files%20(x86)/OPC%20Foundation/UA%201.04/Compliance%20Test%20Tool/help/TestCases/UAServer/All/index.html

The tests with TestType Lab contain a description of the manual steps performed during compliance testing.
e.g. tests from "Security Category\Security Administration" are manual tests with a full description provided.

GregoireG-C, 2020-12-16:
Ok, I see. 
Some test cases are not implemented yet, do we need to to do someting for these? (like Base Information | Base Info Custom Type System).

By the way, you put Kerberos in the test from "Security Category\Security Administration" but it's not a mandatory feature for the certification (standard 2017 opcu server) right ?

closed 2020-12-18 by GregoireG-C