OPCFoundation/UA-.NETStandard #1219: Common practice for Server.Config.xml

DADev99, 2020-12-16:
[ ] Bug [ ] Enhancement [ ] Compliance [X] Question [ ] Help wanted

What is the common practice for the Server.Config.xml file. Is this file usually stored somewhere on the system after installation to allow manual editing, or is it best to hide this file and create an UI to allow configuration of some of the settings.

mregen, 2021-01-07:
Hi @DADev99, the configuration should be stored somewhere with access control, so only an admin or elevated user can edit, similar to the cert store. There are settings in the config which would allows lowering security gates (e.g. AutoAccept), or to make it unusable (e.g. change the cert subject, cert store location etc.).
To create a UI to allow only edit of certain values is defintely a good idea..

mregen, 2021-01-12:
please reopen if not answered...

closed 2021-01-12 by mregen