OPCFoundation/UA-.NETStandard #1223: Skipped test

GregoireG-C, 2020-12-18:
**Type of Issue**
[ ] Bug [ ] Enhancement [X] Compliance [X] Question [ ] Help wanted

**Describe the Issue**

When a test case is skipped, it validates anyways the conformance unit or not ?

mregen, 2020-12-18:
@GregoireG-C could you provide some context regarding the skipped test? Do you mean in the CTT?

GregoireG-C, 2020-12-18:
Yes, indeed.

For instance, in Monitor Item Services / Monitor Value Change, some test cases are skipped because my server doesn't support some specific data types. (but I have others tests skipped).

I am wondering if "skipped" means that my server can"t be tested in this feature but this feature is not necessary for the certification or "skipped" means this test case can't be done (wrong configuration) and so the test case can't be validated.

Regards

mregen, 2020-12-18:
@GregoireG-C I cannot speak for compliance but from my experience the compliance lab decides case by case if a test can be skipped or if its possible to get a waiver. To find out why a test is skipped you could check the test source code. Sometimes because a variable is missing and/or functionality.

GregoireG-C, 2020-12-18:
Thanks for your answer. Have a nice week end

closed 2020-12-18 by GregoireG-C