OPCFoundation/UA-.NETStandard #1227: Versioning Question: CustomNodeManager2, INodeManager2, etc.

justinConstantTime, 2021-01-04:
Hi,

This is a question regarding the class naming convention among the node manager classes (may apply to others):

If one wants to implement a UA server with a simple node manager, is it safe to assume that to keep the code up to date, once I've completed my first implementation of the correct (up to date) interface collection...

`INodeManager, INodeFactory, IDisposable`

... subsequent versions of the INodeManager, will be INodeManager2, 3, 4, etc.?

Based on the samples provided, it looks like the `QuickstartNodeManager` is the closest to the starting point that fits my needs, however it implements `INodeManager, INodeIdFactory, IDisposable`, not `INodeManager2`.

Should I implement `INodeManager2`? Are these names temporary until a better solution is implemented or is this a permanent naming convention?

As a note, the file found at this location:
Libraries/Opc.Ua.Server/Diagnostics/**CustomNodeManager.cs**...

...actually contains a class called `CustomNodeManager2`. This may just be that someone did not name the file correctly.

Thanks!

AlinMoldovean, 2021-01-07:
Hi @justinConstantTime ,

The best approach to implement a node manager in a basic UA Server is to use `CustomNodeManager2`, like _ReferenceServer_ sample project.

The _2_ from _CustomNodeManager2_ comes because there is another _CustomNodeManager_ class marked as [Obsolete]

mregen, 2021-01-28:
hopefully answered...

closed 2021-01-28 by mregen

JustinLipps, 2021-06-04:
Yes, sorry for the late reply, that definitely answers my question!