OPCFoundation/UA-.NETStandard #1228: Auto accept untrusted certificates for client does not work

davidguidali, 2021-01-05:
Hi there
I have a problem / question and am wondering what I'm doing wrong. I've created my own simple opcua server and client using your library as a base. 

First I let it run on my PC as two separate processes and everything worked fine, the client was able to connect to the server without any problems.

Afterwards, I tried to put the server into a docker container connect to it with the client, but now the server gave me a **BadUntrustedCertificate** error, since he didn't trust the clients certificate. To fix this, I set the _AutoAcceptUntrustedCertificates_ flag in the server config file to _true_, rebuild the docker image and tried again (I'm just using this for testing reasons so no need for certificate trust). 

The server now accepted my clients certificate, however, the client did not accept the servers certificate and gave me the same **BadUntrustedCertificate**. OK no problem I thought, just set the  _AutoAcceptUntrustedCertificates_ flag of the client config also to _true_. But this time, this change did not fix the error. **Even though the  _AutoAcceptUntrustedCertificates_ is set to true in the clients configuration, the client is still rejecting the servers certificate!**

I found a workaoround by subscribing to the _CertificateValidation_ event of the clients _CertificateValidator_ and setting _e.Accept = true_ inside the event method, but I'm still asking if there is anything else to do to make the _AutoAcceptUntrustedCertificates_ work for the client?

mregen, 2021-01-07:
Hi @davidguidali, this behaviour as you describe it is known since a long time. 
I agree it should be fixed in the client code because its misleading. Keep also in mind it only accepts untrusted certificates which pass all other checks. If you need to suppress other errors, you should consider implementing your own cert callback anyway.

AlinMoldovean, 2021-04-26:
PR #1380 should fix this...

closed 2021-04-26 by AlinMoldovean

singhacadshravan, 2023-05-12:
You can use autoaccept field in the uaclient contructor for auto accepting the untrusted certificates if you want server specific granularity:
UaClient = new UAClient(
				application.ApplicationConfiguration, output, ClientBase.ValidateResponse, _config)
			{
				AutoAccept = autoAccept
			};

mregen, 2023-05-12:
yes, this is fixed

commit f0c46741 (2021-07-09): Implement AutoAcceptUntrustedCertificates flag for Client side also. Fixes #1228 (#1380)