OPCFoundation/UA-.NETStandard #1239: How to Read ExtensionObject Array?

cfzping2019, 2021-01-14:
[image]

I definded a class like the PLC server node struct like this:
`public class Good
    {
        public short Length { get; set; }
        public short Width { get; set; }
        public short Height { get; set; }
    }`

How to Read the value of goods to List<Good>?

AlinMoldovean, 2021-01-15:
@cfzping2019 ,

From the client side you need to browse and read the Value attribute of the nodes from your structure.

cfzping2019, 2021-01-18:
Hi @AlinMoldovean  !
Can you give us an example of how to read/write structure data?
Thanks!

AlinMoldovean, 2021-01-20:
Hi @cfzping2019 ,

Please have a look at the example here:
https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/ebb14daa12eefb38ebd6d104229852723a322e07/Samples/NetCoreComplexClient/Program.cs#L405-L432

cfzping2019, 2021-01-20:
@AlinMoldovean thanks

closed 2021-01-20 by cfzping2019