OPCFoundation/UA-.NETStandard #1243: Best practice to debug this library from a client using nuget package

tukotech, 2021-01-15:
I have a not so simple client that sometimes blocks (traced it to this [line](https://github.com/OPCFoundation/UA-.NETStandard/issues/new?permalink=https%3A%2F%2Fgithub.com%2FOPCFoundation%2FUA-.NETStandard%2Fblob%2F8c216e78f458c22d3f9f660b5494f3c38f4fdcea%2FStack%2FOpc.Ua.Core%2FStack%2FTcp%2FUaSCBinaryClientChannel.cs%23L337).

I wanted to experiment and make modifications to the library but I am not sure what is the best way to do it without removing nuget package and add a reference to projects. Can someone please provide hints how to do this efficiently?

MD-V, 2021-01-15:
You can use the OPCFoundation.NetStandard.Opc.Ua.Symbols nuget package which will reference all the Debug DLLs (compiled in Debug mode)

See [https://devblogs.microsoft.com/nuget/improved-package-debugging-experience-with-the-nuget-org-symbol-server/](https://devblogs.microsoft.com/nuget/improved-package-debugging-experience-with-the-nuget-org-symbol-server/) for more details

tukotech, 2021-01-15:
@MD-V, thanks for the input I really appreciate it. Note that I am doing this already and I can debug the source. 

I feel like i am on the next steps, I wanted to make some changes to the library and test it. Maybe the question is:
a. Should I build nuget package locally? or
b. Change my project so that it references the "project" libraries?

mregen, 2021-01-15:
@tukotech here is a sample csproj how to map projects or nuget packages:
https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/master/Samples/NetCoreComplexClient/NetCoreComplexClient.csproj

tukotech, 2021-01-17:
@mregen, thanks for pointing the sample out. I think that should solve my problem!

closed 2021-01-17 by tukotech