OPCFoundation/UA-.NETStandard #1245: System.IO.FileNotFoundException: Could not load file or assembly 'Opc.Ua.Security.Certificates, Version=1.4.365.0'

JakubJZ, 2021-01-15:
**Type of Issue**
[ ] Bug

**Describe the Issue**
Opc.Ua expects a nuget package that is not included in the version. 
Although I manually added Opc.Ua.Security.Certificates in version 1.4.365.10, it didn't help.
The solution was to downgrade to 1.4.353.15

**To Reproduce**
netcoreapp3.1 + **OPCFoundation.NetStandard.Opc.Ua version 1.4.365.10** (with certificate support)

**Expected behavior**
- Opc.Ua.Core 1.4.365.10 -> Opc.Ua.Security.Certificates 1.4.365.10

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Windows 10
 - Runtime: netcoreapp3.1
 - OPC Package Release Version 1.4.365.10
 - Component: Opc.Ua.Core

mregen, 2021-01-15:
Hi @JakubJZ, its strange to see that it looks for 1.4.365.0. Could you share your csproj for .NET Core project?

mregen, 2021-01-28:
please reopen if there is still an issue...

closed 2021-01-28 by mregen