OPCFoundation/UA-.NETStandard #1247: Is it possible to read Tag meta data via UA client

albrookesplowman, 2021-01-15:
[ ] Bug [ ] Enhancement [ ] Compliance [x] Question [ ] Help wanted

Is it possible to read Tag configuration via the UA client?  For example, the range, deadband etc... of each Tag in the OPC server

Many thanks

AlinMoldovean, 2021-01-20:
Hello @albrookesplowman ,

In case your question is about the analog data type nodes, you can read the properties like _EURange_ or _EngineeringUnits_ by browsing and reading the property nodes attached to the variable.
[image]

albrookesplowman, 2021-01-20:
So is this information readily available from any OPC server?  We have developed an client which connects to a OPC Server via the Matrikon Tunneller but I am trying to see whether we can get this information from the OPC server or whether it has be configured on the client side for our 6000 signals...  I have had a look around the internet but can't find any information about what information is available from an OPC server

AlinMoldovean, 2021-01-20:
@albrookesplowman ,

An OPC UA Server will expose the mandatory properties for different type of AnalogType variable nodes.
Please make sure you are not dealing in your case with a Classical OPC Server instead of an OPC UA Server.

albrookesplowman, 2021-01-20:
Yes, we are dealing with a UA server - the matrikon tunneller wraps a classic server into a UA server - as far as I am aware :)  Do you know the call that we can use to obtain this information on nodes and is there an example in this repository?  Many thanks for all your help ....

AlinMoldovean, 2021-01-20:
@albrookesplowman ,

Having the _NodeId_ of a variable node, you need to do the following operations:
 - Browse the node like in the example here;
https://github.com/OPCFoundation/UA-.NETStandard/blob/32dc6bcaccd1457bc9eb58312206bb6b8dec654c/Applications/ConsoleReferenceClient/UAClient.cs#L324

 - Go through the list of returned _ReferenceDescriptionCollection_ and find the one with _BrowseName_ = the browse name of the expected property, e.g. for Eu Range should be `result.BrowseName == BrowseNames.EURange`
 - Read the Value Attribute of the requested property, like in this example:
  https://github.com/OPCFoundation/UA-.NETStandard/blob/32dc6bcaccd1457bc9eb58312206bb6b8dec654c/Applications/ConsoleReferenceClient/UAClient.cs#L216

ZbynekZ, 2021-01-20:
@AlinMoldoevan, why do you recommend doing the browse and searching through the returned collection? Isn't the TranslateBrowsePathsToNodeIds service designed just to do that, more effectively?

albrookesplowman, 2021-01-20:
@ZbynekZ  - how do I use that one? And I can give them both a try.....

ZbynekZ, 2021-01-20:
For TranslateBrowsePathsToNodeIds , see https://reference.opcfoundation.org/v104/Core/docs/Part4/5.8.4/ . Basically, in your case, you will specify a browse path (https://reference.opcfoundation.org/v104/Core/DataTypes/BrowsePath/ ) which will have the node ID of the variable node as startingNode, and just one element in the relativePath (https://reference.opcfoundation.org/v104/Core/DataTypes/RelativePath/ ) . This single element (https://reference.opcfoundation.org/v104/Core/DataTypes/RelativePathElement/ ) will specify HasProperty as referenceTypeId, isInverse = false, includeSubtypes = true, and targetName will be the QualifiedName of the property you are looking for.

AlinMoldovean, 2021-01-21:
Hi @ZbynekZ ,

Using _TranslateBrowsePathsToNodeIds_  instead of _Browse_ is more efficient indeed.
_Browse_ has the advantage that it gives you all the existing references (including properties) in one call.

albrookesplowman, 2021-01-21:
@ZbynekZ - Would it be something like

          BrowsePathCollection browsePaths = new BrowsePathCollection();
          BrowsePathResultCollection results = new BrowsePathResultCollection();
          DiagnosticInfoCollection diagnostics = new DiagnosticInfoCollection();

          foreach (var r in references)
          {
            BrowsePath browsePath = new BrowsePath
            {
              StartingNode = (NodeId)r.NodeId,
              RelativePath = new RelativePath
              {
                Elements = new RelativePathElementCollection
                {
                  new RelativePathElement
                  {
                    IsInverse = false,
                    IncludeSubtypes = true,
                    TargetName = "EURange"
                  }
                }
              }
            };

            browsePaths.Add(browsePath);
          }

          SessionConnection.OpcUaClientSession.TranslateBrowsePathsToNodeIds(browsePaths.ToArray(), out results, out diagnostics);

ZbynekZ, 2021-01-21:
Yes, in principle, something like that. I do not know whether "(NodeId)r.NodeId" is correct, because I do not know what type is your "references" input, but hopefully that's correct too.

albrookesplowman, 2021-01-21:
@ZbynekZ  - am i right in thinking that I can then use the results array to read the actual values using the ReadValue call?

ZbynekZ, 2021-01-21:
Yes but bot directly - you need to get the node IDs from the results array of TranslateBrowsePathsToNodeIds (they are in the targetId-s), and then construct the arguments for Read with them.

albrookesplowman, 2021-01-25:
@ZbynekZ - I seem to be getting a Bad_NoMatch for a request for **EURange** for the Tag I am after which I assume means that it doesn't exist...   Is this correct?

ZbynekZ, 2021-01-25:
It means that it doesn't exist, if you have specified all arguments correctly and the server doesn't have a bug... I suggest that you check it in some other OPC UA client such as UAExpert.

tukotech, 2021-01-26:
@albrookesplowman if I may go back to the original question which was read the range, engineering units etc. This might be server specific. 

For example, a DCS/SCADA system that implements [FF Specs](https://www.fieldcommgroup.org/foundation-fieldbus-technical-specifications) that implement AI (Analog Input Function) block will have the range and engineering units as part of the function block.  An AI function block may provide the engineering unit like:
AI1/XD_SCALE.UNITS
AI1/XD_SCALE.EU0
AI1/XD_SCALE.EU100

What was the Matrikon connected to?

albrookesplowman, 2021-01-26:
@tukotech - many thanks for the information.  At the moment we are using the Matrikon UA Tunneller to wrapper around a OPC classic server so that we can connect to it as a UA client.  We are developing a system that will either connect direct to a UA server or to other servers using the Matrikon Tunneller software.  I do not currently have any further information about the OPC server that is at the other end of the tunnel but I will try and find out.  We are about to deploy a test around a OPC DA server (again via the Tunneller software) so I will be able to do some further tests with that too.  

Do you know how to read the information if the DCS implements FF Specs?  Is there a means of getting this information?

tukotech, 2021-02-01:
@albrookesplowman I only have access to Emerson's DeltaV DCS and yes DeltaV exposes data via ISA-95 equipment model and the control modules supports FF specs model.

closed 2021-05-25 by AlinMoldovean