OPCFoundation/UA-.NETStandard #1253: Reverse Connect

charlesaugustineabb, 2021-01-20:
**Type of Issue**
 Question 

**Describe the Issue**

OPC Foundation UA SDK (Server and Client )Supports "Reverse connect" functionality implementation?

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Sample '...'
2. Connect to Server '....'
3. Browse Address space '....'
3. Subscribe to Node '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log output**
If applicable, add log output to help explain your problem.

**Code snippets**
If applicable, add a code snippet to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10, Ubuntu, macOS]
 - Development environment: [e.g. Visual Studio 2019 16.7.2]
 - Runtime: [e.g. .NET 4.6.2, .NET Core 3.1]
 - OPC Package Release Version [e.g. 1.4.363.107]
 - Component: [e.g. Opc.Ua.Core]
 - Server: [e.g. Reference Server]
 - Client: [e.g. UA Expert]

**Additional context**
Add any other context about the problem here.

mregen, 2021-01-20:
Hi @charlesaugustineabb, short answer is yes. see --> https://github.com/OPCFoundation/UA-.NETStandard/blob/master/Docs/ReverseConnect.md

charlesaugustineabb, 2021-01-25:
Thank a lot mregen for quick reply.

closed 2021-01-25 by charlesaugustineabb