OPCFoundation/UA-.NETStandard #1256: Logging and tracemasks - Question/Enhancement

pgrimmert, 2021-01-22:
**Type of Issue**
[ ] Bug [X ] Enhancement [ ] Compliance [X] Question [ ] Help wanted

**Describe the Issue**

Current state of opc tracing:
The implementation of the logging functionality is a global accessible tracing class instance.
TraceMasks is implemented as "public static class TraceMasks" with having const int with values set as hex.
TraceOutput is implemented as enum "public enum TraceOutput" with int values.

Question:
I wonder if there are any plans to implement a more sophisticated logging approach by injecting ILogger, which would help controlling what we want to log. 
Also is there any specific reason why the class TraceMasks is used like an Enum with bitflags but is not actually an enum with bitflags? in the same class, the TraceOutput is implemented as enum. 
Its confusing.

Also the current implementation does not allow the possibility to host multiple OPC servers within the same program instance e.g. as hosted services. Since they share the core code which does also make use of the Util.Tracing.

Any feedback is welcome. 

Regards
Philipp

mregen, 2021-01-28:
@pgrimmert , this is a hot topic, I started #1262 to collect feedback for the feature work. Please add your main concerns, requests suggestions to #1262.

mregen, 2021-12-16:
see #1631

closed 2021-12-16 by mregen