OPCFoundation/UA-.NETStandard #1258: Stack overflow when subscribing to all nodes under Server object

kimaluco17, 2021-01-22:
**Type of Issue**
[x] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

**Describe the Issue**

When I run the ReferenceServer and connect to it using UA Expert, if I add the Server node to the Data Access View a StackOverflowException is sometimes thrown. If that succeeds, deleting all the items from the Data Access View also results in a StackOverflowException. This is apparent in both the .NET Core and the .NET Framework sample servers.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile Sample, run ReferenceServer
2. Connect to ReferenceServer using UA Expert
3. Add Server node to Data Access View for subscribing/creating monitored items - sometimes throws exception
4. Delete all monitored items from Data Access View - always throws exception

**Expected behavior**
No exception should be thrown.

**Screenshots**
Call stack from exception thrown on add:
[image]

Beginning of call stack:
[image]

**Environment (please complete the following information):**
 - OS: Windows 10
 - Development environment: VS 2019 Professional 16.8.3
 - Runtime: .NET Core 3.1
 - OPC Package Release Version: 1.4.363.107, 1.4.365.22 (master branch)
 - Component: Opc.Ua.Core
 - Server: ConsoleReferenceServer, QuickstartsReferenceServer
 - Client: UA Expert

AlinMoldovean, 2021-01-25:
Hello @kimaluco17 ,

We will investigate this and report the results.

AlinMoldovean, 2021-01-26:
From the call stack it seems to be a loop generated by _DiagnosticsNodeManager.DoScan(object alwaysUpdateArrays)_ started by a _DiagnosticsNodeManager.OnMonitoredItemDeleted()_

I tried to replicate the replicate the StackOverflow exception with the provided steps without success.
I used the source code from the master branch and UA Expert version 1.5.1.

@kimaluco17, could you provide some more details to replicate the error?
Do you have any changes in your ReferenceServer code or configuration?

kimaluco17, 2021-01-26:
@AlinMoldovean 

Thanks for looking into that, I have not changed anything in the repo, using the default configuration, and also using UA Expert 1.5.1 with default configurations besides the TypeDictionaryMode being set to off. I can also reproduce this from another application that is under development, where we are using the NuGet package version 1.4.363.107.

I will check with my teammates to see if they run into the same issue, sounds like it could be something to do with my environment.

AlinMoldovean, 2021-02-03:
@kimaluco17 ,

Could you please check if the fix from  #1258 fixes the issue?
You can test it with the latest version from `master `branch.

closed 2021-02-03 by AlinMoldovean

kimaluco17, 2021-02-03:
@AlinMoldovean 
The changes in #1270 did fix the issue. I also checked with a teammate, he was not able to reproduce prior to that fix so it does seem like an environment related issue.

PR #1270 (2021-02-03): Avoid loop in DiagnosticsNodeManager.OnReadDiagnosticsArray() .Fixes #1258
Avoid DoScan() loop in DiagnosticsNodeManager.OnReadDiagnosticsArray 
Fixes #1258
commit c14d84ce (2021-02-03): Avoid DoScan() loop in DiagnosticsNodeManager.OnReadDiagnosticsArray() . Fixes #1258 (#1270)