OPCFoundation/UA-.NETStandard #1262: Discussion, request for feedback: Future logging experience in the UA stack

mregen, 2021-01-28:
**Type of Issue**
[ ] Bug [X] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

**Describe the Issue**

Default logging is currently a very limited experience and has perf issues when completely enabled.
There is a way to plug in a trace handler, but still the log messages are sometimes too many (serevr monitored items)
sometimes too fees.

The ask is here for all contributors, users of the UA stack to provide feedback to be able to come up with a comprehensive feature list.

**Expected behavior**

- pluggable logger to allow to use serilog and the likes
- logging zones to allow to switch logging areas for say client, security, tcp.
- log levels, mapping?
- dynamic configuration for logs
- best performance, no impact on client/server perf

Just a starter here, please provide feedback.

BrunoJuchli, 2021-01-29:
We sometimes create multiple concurrent sessions in a single process to connect to different kinds of devices.
When reading the trace file it's currently very hard to distinguish which entry stems from which session.
There should be a mechanism "fixing" this. Several come to mind:

- be able to specify a separate log file per session
- be able to pass an identifier to the session which the session puts in the log (in a separate log "column" so we can filter for it)
- pass the responsibility of writing log files on to the consumer of the library (preferred!)

Let me elaborate on the last point - this also connects to logging with existing logging libraries:
As per experience from other projects I'd recommend against linking to specific logging frameworks or abstractions.
Instead, define your own logger interface, like for example:

    public interface IOpcUaLogger
    {
        void LogError(string message, Exception ex);
        void LogError(string message);
        void LogInfo(string message);
        ....
    }

the consumer of the library has then to implement that logger -- usually that will just be a wrapper for the logging library he uses, for example:

    public OpcUaLog4NetAdapter : IOpcUaLogger
    {
        private readonly Log4Net.Logger logger;

        public OpcUaLog4NetAdapter(Log4Net.Logger logger)
        {
            this.logger = logger;
        }

        public void LogError(string message, Exception ex)
            => this.logger.LogError(message, ex);

        ....
    }

--> creating our own logger-adapter is trivial but frees us from dependency constraints which otherwise always lead to problems down the line. It also allows us to give content to the log. In the above example, the log Log4Net logger could be created with a loggername like "Thermometer[Room5].OpcUaSession". No need for Ua-Foundation to know where it's logging to but I'd still be able to filter my log for the opc ua session of room 5 (not room 4, not room 3...). It's all in my control so no worries for you!

The IOpcUaLogger should then be passed to the relevant parts of the OpcUaLibrary, for example:

    new Opc.Ua.Client.Session(..., logger)

Note: for backwards compatibility reasons the logger could be optional -- fallback to the current approach of a single trace log file.
Personal opinion: I don't care much for backwards compatibility in this regard (neither in lots of others...). Release notes should describe the change and give a description of what's to change, that would make me very happy already.

---------------

Regarding log levels and configuring logging levels for specific areas I haven't given much thought.
Seems to me though when you give us control about logging then we can use "out of the box" logging-library features to switch  logging on/off. Good for you, because you don't have to worry about it. Good for us, because it integrates with the tools we already use.

mregen, 2021-01-30:
I added a sample how to use the trace instance for external logger (serilog) https://github.com/OPCFoundation/UA-.NETStandard/pull/1263

ZbynekZ, 2021-02-03:
By properly setting the parameters and hooking to Tracing.Instance.TraceEventHandler, it is possible to direct all tracing to your own code and, if you like, to any kind of tracing/logging infrastructure. I am already doing that in my own library.

I think that all that remains to fulfill the various needs expressed here is a richer content of the trace event, which can be done by extending the TraceEventArgs (or even deriving more specialized events from it; but rather not). The extra content can include the category/zone, level, session reference, etc. All of that can be introduced gradually inside the stack/SDK, without breaking the code compatibility. Further filtering and directing the trace events can be left to the consuming code.

For good performance (and also better opportunity for further event analysis), where possible, formatting of the messages should not be done on the producing side, but left to the consumer. There is, however, a provision for it inside the TraceEventArgs as well  (format + object[] args),

fremat, 2021-02-10:
As a consumer of OpcUa I think Microsoft.Extensions.Logging.Abstractions is a good abstraction to use for logging,
The Opc.Ua.Bindings.Https already have a reference to it by using Kestrel.
It's extensively used in open source projects.
It's pluggable, most loggers already have extensions for it.
Low overhead if used correctly.

The consumer of OpcUa can freely choose the logging framework to use.

You can name your loggers (logging zones), have standard log levels.
The consumer can set up filters individually for log providers.

ZbynekZ, 2021-02-10:
@fremat You are right that Microsoft.Extensions.Logging.Abstractions is a good abstraction, but in my view it is unnecessary to introduce a dependency on that. Because if somebody wanted to use a different framework (Log4Net, Serilog, NLog,..), he would have to hook it into Microsoft.Extensions.Logging.Abstractions. Which is only a bit easier (or not) than hooking into the existing Tracing.Instance.TraceEventHandler (see my post), except that we now have extra dependency and an extra layer for the calls to go through.

fremat, 2021-02-11:
@ZbynekZ In my view, there already is a dependency on Logging.Abstractions and its used by over 3.3K projects (nuget.org data). It has a _very_ high adoption rate.
In our internal projects we pull logging events from other frameworks such as AspNetCore, EntityFrameworkCore, MassTransit (AMQP) etc. from Logging.Abstractions, it works really well, no extra hooks. All logging configuration is consistent and consolidated.

Tracing.Instance.TraceEventHandler looks like a proprietary solution. Its not complicated, but you need to look it up, and create your own hooks for it if you want to have OpcUa logging setup in your application.
In its current form it's also not composable. It uses a singleton pattern on concrete types. Moving to an abstraction allows for future enhancements.

Also, in my view, logging and tracing are not the same, but they overlap.
For high perf. tracing in hot paths it might be better to use System.Diagnostics.Tracing.EventSource.
You can get high volume traces to ETW in Windows or LLTng framework on Linux.
It shouldn't affect performance in production, only when you hook up EventSource listeners.
And you can also setup your own EventListeners for those. if you need it in process.

ZbynekZ, 2021-02-11:
@fremat: 

My view is as follows:
1. It is already possible to hook any kind of logging/tracing into UA-.NETStandard.
2. The information currently provided in the UA-.NETStandard tracing facility is, however, insufficient. It does not give you event IDs, categories/zones, session references etc.

To me, what you are suggesting seems to be trying to resolve point 1, which is not a problem at all. And it is ignoring point 2, which is what we really need to solve.

And, if Logging.Abstractions is adopted, it would most likely not be done by changing every place where the current tracing already appears. That would be quite a work, and may break code compatibility. So what you will end up doing would be hooking the Logging.Abstractions into some central place through which all current tracing goes through - and that will be Tracing.Instance.TraceEventHandler or similar place. And because the main thing in Logging.Abstractions is the ILogger.Log<TState> method, you would use the concrete type that we already have, TraceEventArgs, as TState. That is almost as simple as one line of code. And that's how your solution will look like. What I am saying that it is absolutely unnecessary to do this in the library, because it is so simple that anybody can do it.

I want also want to address some more concrete points:

You wrote "... there already is a dependency on Logging.Abstractions": Not really. What you are describing is that some internal package that we indirectly depend on uses Logging.Abstractions". That is, however, an implementation dependency that we do not have to care about, and may change. But we do *not* have dependency on Logging.Abstractions of our own API that we provide to developers, yet. And that's what you suggest to introduce. 

You wrote "... logging and tracing are not the same, but they overlap.": I definitely agree. And that is one more reason not to use Logging.Abstractions, because that is more suitable for high-level logging, while all or vast majority of the information that currently comes from UA-.NETStandard is more like "tracing". 

All in all, I can see your point - you are trying to "standardize" the library, by having it use a popular interface. But my position is different - I want the UA-.NETStandard only contain what is truly necessary, and let everything else be done by the consumers.

My proposal is to let this part ("how to hook into the library" - point 1 above) unchanged, and focus instead on Point 2 above, i.e. extending TraceEventArgs and then changing the places where trace events are generated so that they feed in the right information (and possibly adding new places).

Kind regards

BrunoJuchli, 2021-02-18:
> I added a sample how to use the trace instance for external logger (serilog) #1263

Thanks again for the sample. I now had a more thorough look at it, as far as I can see logging extensibility boils down to the existence of `Utils.Tracing.TraceEventHandler` (as seen [here in the example](https://github.com/OPCFoundation/UA-.NETStandard/blob/8c9e722dfe054f71196192daf47e2055d1ee1217/Applications/ReferenceServer/SerilogTraceLogger.cs#L93)).
(I'm explicitly stating this because it might help someone else to not have to go through all the code changes of the PR (mostly style changes and thus noise in regards to logging).)

Problem with this is, sorry if I'm overly repeating the topic, that it is unsuitable for separating log output for multiple sessions (also see #594 and #1256).
I'd very much like the ability to specify a log-handler per session.
I know it's not quite *that* simple because there's also logging done before a session is actually created.

I suggest to redesign by having the logging-handler be injected into every component which performs logging instead of using a static (global) logging infrastructure.

--------

Note: in case it's not obvious why the static log-handler is unsuitable to separate log output for multiple sessions I'll happily elaborate further.

ZbynekZ, 2021-02-21:
@BrunoJuchli You suggest injecting a log handler into every component which performs logging. I do not see how this helps. There is large number of these "components" (various objects related to communication), and you (as developer using UA-.NETStandard) do not create most of them. They get created by other objects as need arises. So, you do not get a chance to inject your own logging handler into them, unless you rewrite the whole library into more IoC/DI -like approach. The object inside the library which creates a new object, not knowing anything better, will have to pass it own logging handler into the newly created object, so you end up with one single handler being used again, except that you have spent extra time on working on it, and increased the code size.

What is really missing is an information inside the trace events that are currently being generated - "correlation", "activity", or however you would like to call that. It has been mentioned here as missing "session" information, which is the case in many times, but at other places it might be something different. This information is missing in the lower-level objects (the channel/socket objects do not know which UA session they are serving, for example). So you need a) pass this information from the upper-level object to lower-level objects, and b) provide that information in generated trace events - which can be done by improving the individual places where they are generated, and expanding TraceEventArgs.

As it is clear from what I wrote, it is not obvious (at least to me) why the static log handler is unsuitable to separate log output for multiple sessions, so can you please elaborate on that further.

Kind regards

BrunoJuchli, 2021-03-02:
@ZbynekZ 
It's very well possible that this requires some refactoring of the library.
Given that refactoring is performed, your argument about one single handler being used is void.

I don't think it's impossible to achieve the goals using a static handlers - I just think it's significantly harder (for the consuemr) and thus unsuitable.
Let me make an example:

You've got two different components in one software which use OpcUa, lets call them "machine" and "robot".
If you use correlation ids (session id,.. what ever) I either have to implement an object oriented logging adapter which then uses the correlation id's to map the log events to my machine and robot logger objects.
Or my machine and robot implementations need to listen to all logging events but then filter for correlation ids. How many relevant correlation ids are there?

Now if instead I can inject a logger interface the correlation is given by that logger object - plain and simple.

You're not convinced yet? Let me address your specific comments:

>You suggest injecting a log handler into every component which performs logging. I do not see how this helps. They get created by other objects as need arises. So, you do not get a chance to inject your own logging handler into them, unless you rewrite the whole library into more IoC/DI -like approach. The object inside the library which creates a new object, not knowing anything better, will have to pass it own logging handler into the newly created object, so you end up with one single handler being used again, except that you have spent extra time on working on it, and increased the code size.

How is this different from the following?:
>This information is missing in the lower-level objects (the channel/socket objects do not know which UA session they are serving, for example). So you need a) pass this information from the upper-level object to lower-level objects,

I can imagine two possibilities how correlation ids are generated and passed around:

- the correlation ids are autonomously generated by the library. But then how would I map these ids to my robot and machine components? I simply could not - at least not without creating something as least as complicated as the logger-injection concept (you're welcome to prove me wrong).
- the user of the library generates the correlation ids and passes them in. But then, how is that any simpler than passing in a logger interface? The information (correlation id or logger interface) both need to be passed around to be made available when logging and they have the exact same scope. --> not simpler at all.

Note: it may still make sense to make correlation ids defined by the protocol available - if there any like that. For example, if the OpcUa protocol declares the each session has an id, then this id should be made available when logging. How this may work depends on the specification of the protocol, for example when a session-id is only available upon successfully establishing a session, but the session object may be created beforehand, then it's obvious that the session id is not available when injecting the logger into the session. Connection failure may thus not contain the session id. The session id is made available only after successful establishing of the session. This might be resolved by making it part log information, in the simplest case part of the message, alternatively it can be passed as argument to the logger. Example:

    ISessionLogger
    {
        Warn(int sessionId, string message, object[] arguments);
        ...
     }

ZbynekZ, 2021-03-02:
@BrunoJochli thanks for reply. I can see now that the different views come mainly from 1) different requirements, and 2) different understanding of what "logging" or "tracing" is. More detail:

1) I do not have such requirement that you described - that is, to effectively separate those different "components" using the library. I am not sure how common it is, so it would be good to know from others here what their views are.

And, related to it,

2) I would not, in effect, called the approach you describe neither "logging" or "tracing". If you can, from outside, specify on deep levels both the actions that happen, AND correlation data, it is conceptually a callback mechanism for various events inside the library. My understanding of usual meaning of both "logging" and "tracing" is that this is something that they are intentionally not designed to do - they are meant to be primarily built-in mechanism that you can hook into for processing the events, but still on a more or less global level. 

To the less important points:

A)
> How is this different from the following?: ...
 
In your approach not only some already existing correlation data need to be passed to created objects (such as from session to subscription, or to communication objects), but also some mechanism for obtaining the logging interfaces and your own correlation data for deeper levels needs to be put in place.

B) In the last example with session logger, if goes on referred to further, please do not use "session Id" for the correlation parameter you are introducing, even if it is an 'int', because Session Id is a term with a defined (different) meaning in OPC UA. 

To make it clear, in the approach that I suggest, there might be such correlation parameter for the session too, but it would certainly be created by the library itself, and not passed in externally.

All in all I believe we will now probably have good understanding of the different requirements and suggested approaches, so the question really remains what the prevailing requirements are (and whether somebody will actually be willing to do the coding work for either of them...).

Best regards

BrunoJuchli, 2021-03-03:
@ZbynekZ 
Thanks for your comment and your interest.

I consider this a rather basic logging feature. Basic enough, that its supported by most logging libraries and even log viewers.
Here a (simple!) example of one of our logs where we have the same kind of component twice but have separated log output (Magazine/P1 and Magazine/P2).

[image]

We sometimes have many more components of the same kind (8 machine, 6 magazines,...).
Each of those has their own OpcUa connection/session.
If we then have an issue with a specific component it would be very cumbersome to have to look at the OpcUa logs of all the components instead of just the affected component.

Also notice the context-aware logging functions of logging frameworks. Examples:

- [log4net](https://logging.apache.org/log4net/release/manual/contexts.html)
- [NLog](https://github.com/NLog/NLog/blob/9169a9f7e07ecb376c725ccacea332d8e9ef89aa/tests/NLog.UnitTests/LayoutRenderers/Contexts/NDCTests.cs#L98)
- [Serilog](https://nblumhardt.com/2016/08/context-and-correlation-structured-logging-concepts-in-net-5/)

Also: logging frameworks are usually designed for object oriented (context included in object) not procedural (context is being passed around as data):

- `log4net.LogManager.GetLogger("context")` --> returns logger object
- `NLog.LogManager.GetLogger("A")` --> returns logger object
- serilog: `Log.ForContext(...)` --> returns logger object

mregen, 2021-09-07:
Hi folks,
I started some prototyping of a new core logging engine, but I think it is not for sure all of the requirements  can be met.
Just recollecting a few here:
- API: backward compatible with `Utils.Trace` messages on app side.
- API: maintain `TraceMask` as best effort.
- API: maintain the `TraceEventHandler` compatibility and usage pattern.
- avoid dependency on logger other packages
- goal: run apps by default with trace level without affecting runtime behaviour even in production so logs give a decent understanding of events, not only errors.
- goal: be able to debug high frequency events and perf, stretch goal even in production 
- goal: log or filter for seperate sessions or area of interest 
- built in file writer should be deprecated due to poor perf and overhead

The problem in the current codebase is that 
- ~250 references to Utils.Trace with unclassified log level. Some of it with a very high frequency, some really informational 
-- ~200 of which are with format/parms
-- all of these messages must be triaged to map to appropriate level or area of interest and/or need to be improved for better information messages
- ~200 refs to Utils.Trace for exceptions, those are typically rarely happening so ok, but mostly there is only little context.
- other trace messages tied to debug or commented out
- `Tracemask` is rarely used with multiple flags, mostly just `Information`, `Security` etc., that makes it easier to map

As a core logging engine the recommendation I got is to use an `EventSource`. It is the core engine also in .NET and is recommended for libraries. It allows to hook up well known windows perf tools like `perfview`, but is also supported on linux using `dotnet-trace`. It can be used to have a minimal impact on perf even when enabled, so all high frequency events can be logged and debugged.
An `EventListener` builds the bridge to the backward compatibility layer here, e.g. the apps which simply hook up to the `TraceEventHandler` can do so if required. Applications may also choose to implement their own `EventListener`.
But it comes with a few caveats. E.g. Etw only supports a limited set of datatypes, int, DateTime and string.

To build a decent EventSource the approach I'm looking at is the following:
- split warning level aka tracemask at EventSource message level to seperate methods 
- low frequency messages with parms are formatted at runtime.
- high frequency messages get their own method with metadata.
This approach allows with a limited effort to support most scenarios above.

Backcompat for apps:
- introduce a new IOpcUaLogger interface in `Utils` with methods for warning levels, similar to other `ILogger` interfaces from serilog and the likes.
- new Trace/Log call is something along the lines of: `Utils.Log.Warning("This is a warning code {0}", StatusCodes.BadInvalidArg)`
- Utils.Trace messages map to the new IOpcUaLogger interface for back compat.
- The new `Utils.Log` implementation maps to an `EventSource`, however, for customization it should be allowed to replace it, similar to the `TraceInstance` mechanism.

problems to solve:
- the TraceEventHandler gets as parameter the format/args variables, so calling it from the `EventListener` is only possible with already formatted strings, which kills perf for that type of logging support. 
-- Solution: i) callback `TraceEventhandler` from `EventSource`. ii) only support the TraceEventHandler as long as no other EventListener is active.
- invent 'correlation id' which allow to filter for session or subscription or e.g. networking?
- How to filter for message source (e.g. is it a client/subscription type of message)
- Should there be multiple EventSource or just a single one? Multiple event source may allow to enable tracing
- Audit logging: Since this is going to a log file, there should be a seperate call to log intrusion attempty, bad certs errors etc.

This is my current thinking, please share feedback. My working branch is [here](https://github.com/OPCFoundation/UA-.NETStandard/tree/logging) but not quite ready for discussion.

mregen, 2021-12-09:
Hi @BrunoJuchli , @ZbynekZ , @fremat , please check #1631 .

closed 2021-12-15 by mregen