OPCFoundation/UA-.NETStandard #1280: Need some clarity on Discovery Mechanism

RakshanPremsagar, 2021-02-08:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [x] Question [ ] Help wanted

Well, not an issue exactly, but need some clarity on the discovery mechanism.
I have a custom LDS from the open62541 stack which is mDNS enabled. Next i register another server to it, so the additional server which is registered is identified in the network. 
I am emulating a scenario of a unplanned/unforeseen/ungraceful shutdown. So, i close the running instance of the terminal of the addtional server, leaving on LDS running. Theory promises a default time duration of 10mins, after which the server which was abruptly closed should not be available. However, what I observe is that - both the servers are shown in the network for as long as the LDS is alive.
Is it a theoretical misunderstanding that I have got or is it the case always?

(The deregistration of the server should happen from the server side and LDS has no way to deregister the servers which are already registered. Even if I put a additional heartbeat from the LDS to the server, de registration API should always initiate from the server side, and the heartbeat scneario would not be useful. Any guidance is welcome..)

AlinMoldovean, 2021-02-08:
Hello @RakshanPremsagar ,

An LDS implementation should have a mechanism to clean the abandoned server registrations, i.e. if a server does not renew its registration within a specific time frame, it should be unregistered automatically by LDS.

The [official LDS](https://github.com/OPCFoundation/UA-LDS) implementation of OPC Foundation has a configuration flag for this:

```
# maximum age of registered server entries before they expire in seconds. (Default=600)
ExpirationMaxAge = 600
```

RakshanPremsagar, 2021-02-09:
Hello @AlinMoldovean ,
 thank you for your response.
If it is a case that we use LDS with multicast enabled, the advertisements of the server which is registered to the LDS will be present inspite of the ExpirationMaxAge right? To completely remove them off, the registered server has to be shutdown.
But while experiencing a unwanted shutdown, the LDS has to take the initiative to shut or remove whatever entries which were being generated.

Also, the parameter ExpirationMaxAge. does it work throughout the network ( I mean if you are checking for servers from another host(findServersOnNetwork))? or it is confined to results on the same host itself? (findServers)

AlinMoldovean, 2021-02-09:
Hi @RakshanPremsagar ,

LDS can remove a server from the local registration list (when registration time expires) and stop announcing it as available through mDNS.
This topic should be addressed on the specific implementation you are targeting.

closed 2021-02-16 by RakshanPremsagar