OPCFoundation/UA-.NETStandard #1282: CertificateValidation Event is no longer called

ReIaxo, 2021-02-09:
From Version 1.4.365.10 and newer we got problems accepting Server Certificates, cause the Event is not fired. With Version 1.4.365.2 everything works fine. In the changelog we saw some changes affecting the validation. Do you changed the way we have to handle it or is it a bug now?

We do it with 
`applicationConfiguration.CertificateValidator.CertificateValidation += MyMethod(CertificateValidator validator, CertificateValidationEventArgs e);`

We read some other issues where people refer to "Listening to the validation callback". Is it what we do or is it another method do archive the goal of accepting a server certificate?

If we do it wrong, please tell us how to do it right. We already looked through the sample applications, but they are also use Version 1.4.365.2. 
Thank You.

mregen, 2021-02-12:
Hi @ReIaxo , there is a change the way the certificate validator reports errors, but it was designed to be compatible with existing applications. 
In the past the cert validator reproted only the first suppressible error, now it may call multiple times if multiple errors must be approved. There is a new AcceptAll variable to bypass other callbacks, if the app is able to inspect the chain.

Sample is here: https://github.com/OPCFoundation/UA-.NETStandard-Samples/pull/120
the dialogs can show all suppressible errors.

Also to simplify domain checks it is now also reported by the callback.

If you have a specific issue which stopped working please report..

ReIaxo, 2021-02-24:
Hi, it now works fine for me. It was about my nuGet package references. They where kinda messy. I cleaned them up and now the Certificate Validation Event is called. Thank you.

closed 2021-02-24 by ReIaxo