OPCFoundation/UA-.NETStandard #1288: Question on GDS

RakshanPremsagar, 2021-02-16:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [x] Question [ ] Help wanted

Hi there,
I am trying to get clarity on OPC UA Discovery mechanism.  In multicast subnet discovery, the scenario for not knowing which hosts have servers could be handled by the use of MDNS enabled Local discovery servers.
Specification says, the use of GDS can be extended across administrative domain. Does this mean, the GDS provides mechanism to find the servers across multiple networks? Also, the specification 12, in figure 7 describes the use of GDS within the same network. What could be the advantage of it, when there is a LDSME doing the same?

Can GDS be seen as the alternative when LDSME is not available?
I figure that GDS encorporates a certificate management system, which has certficates signed by the CA for all the applications registered to the GDS, facilitating the removal of manual exchange of certificates between the client-server communication.

Can the GDS implementation of the stack, be used just for discovery purpose and not security management? From what i understand the facility of Discovery and Certificate management are two independent processes?

Please let  me know. Looking forward for some clarity here.

Thanks in advance,
Rakshan

mregen, 2021-09-14:
Hi @RakshanPremsagar, at this point the spec states that mDNS support is only available in a LDS. So the LDS exposes the applications on a local server, or with mDNS support on a local subnet. The GDS can be used as discovery on a factory/company level. Because the registration of a client/server requires admin rights, the automatic discovery using mDNS is not supported. 
A LDS and GDS serve a different purpose, so they coexist side by side. Hope this helps.

mregen, 2021-09-14:
Please reopen if necessary.

closed 2021-09-14 by mregen