OPCFoundation/UA-.NETStandard #1298: Is it possible to read historical data via UA

albrookesplowman, 2021-02-25:
Hi, 

I apologise for all the questions about UA but it is an area that I am slowly getting up to speed on - it is a very big topic :)

I have a need to be able to read historical data from our OPC server.  Is it possible to read data from an OPC server via an UA client?  Is it simply a matter on connecting to the port for the historian and subscribing for data in a similar way to normal.  If someone update data in the historian, does a UA client get a notification/event of this change.

Many thanks to all the people in this community for all their help so far...

Alan

AlinMoldovean, 2021-02-25:
Hello @albrookesplowman,

A Client/Server sample project for Historical Data Access implementation can be found in [UA-.NETStandard-Samples](https://github.com/OPCFoundation/UA-.NETStandard-Samples) repository:
https://github.com/OPCFoundation/UA-.NETStandard-Samples/tree/master/Workshop/HistoricalAccess

The project is included in [UA Quickstart Applications.sln](https://github.com/OPCFoundation/UA-.NETStandard-Samples/blob/master/UA%20Quickstart%20Applications.sln)

albrookesplowman, 2021-03-10:
@AlinMoldovean - thanks for the info... I am having a look at the Historical Access workshop and I have found the historical events workshop too but i haven't ran that one yet.  Is there any documentation to highlight what you are meant to be able to do with these workshop projects?  for example, the server/client/aggregator tester.... Is it possible to edit the data in the historical server using the aggregator and then read the data via the client?

albrookesplowman, 2021-03-11:
Now that I have had a look at the historical access and events workshop projects, I am slightly confused as to what is possible.  In simple terms, I would like to be able to read the historical values for a particular node/tag and then get notification when a value changes for this node/tag...  Is this possible and do you know which section of code would be a good place to look?  Many thanks

AlinMoldovean, 2021-03-15:
@albrookesplowman ,

Reading of historical values and receiving data change notifications are separate operations in OPC UA.
For receiving value change notifications you need to create MonitoredItems for required nodes.
For reading historical values you need to call HistoryRead for nodes with history support. _HistoricalAccess_ sample provides an implementation example, but it does not have a detailed documentation.

closed 2021-03-23 by AlinMoldovean