OPCFoundation/UA-.NETStandard #1301: ServiceResultException: BadOutOfMemory

JuYanYan, 2021-02-26:
I uses .Net 5.0 with Visual Studio 2019 on Windows, then, when I try:

```c#
Session session = Session.Create(
                        cfg,
                        endpoint,
                        false,
                        false,
                        cfg.ApplicationName,
                        30 * 60 * 1000,
                        new UserIdentity(),
                        null
                    ).Result;
```

It throwed exception:

> ServiceResultException: BadOutOfMemory

However, in configure (a xml file) file:

```xml
....
        <AutoAcceptUntrustedCertificates>false</AutoAcceptUntrustedCertificates>
        
        <RejectSHA1SignedCertificates>false</RejectSHA1SignedCertificates>
        <MinimumCertificateKeySize>1024</MinimumCertificateKeySize>
        
    </SecurityConfiguration>
....
```

Only 1024 bytes.

And, when I delete node "MinimumCertificateKeySize"(or let it equ 2048), it will throw:

> ServiceResultException: Certificate doesn't meet minimum key length requirement.

Can anybody help me?

Thanks

mregen, 2021-02-27:
@JuYanYan could you provide the callstack of the BadOutOfMemory exception?
The second exception is hit when the server you are connecting to uses a deprecated security policy with a 1k key cert, e.g. Basic128Rsa15. Testing with some newer server or the reference server in this repo may help to determine the root cause of the issue.

JuYanYan, 2021-03-23:
I'm very sorry. I've changed another solution. I'll find a free time to repeat this problem.

The call stack is small, like this:

```c#
ConnectOPCServer
Main
```

I cannot connect NI OPC Server (2016), and, I used KEPServerEX 6 and caused ServiceResultException: BadOutOfMemory.

May be I should try with the reference server.

closed 2021-04-27 by AlinMoldovean