OPCFoundation/UA-.NETStandard #1302: Write Node Value with AccessLevel CurrentRead

JayDalton, 2021-03-01:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [x] Question [ ] Help wanted

**Describe the Issue**

Hello,
i am using this library in an C++/CLI environment for both, custom server and client application.
Referencing to a closed question from 2019 [#850] i am wondering why i could not write to a variable with UserAccessLevel =CurrentRead. Following the instructions, i can not get success.

**To Reproduce**
Steps to reproduce the behavior:

So i tried the example in "SampleApplications/Workshop/UserAuthentication" and set some Breakpoints.
I connected me [windows credentials] with an custom client successfully and try to write a new value. I hit the EventHandler "OnReadUserAccessLevel" and raise my AccessLevel. But for some reason, i never hit the EventHandler "OnWriteValue". 

**Expected behavior**
I expected to hit the EventHandler "OnWriteValue" or something else the make my request happend. But it fails to write a new value  with message: BadUserAccessDenied. 

Is this the right approach for writing values with reduced AccessLevel? 
Do I have more opportunities?

**Environment (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Development environment: [e.g. Visual Studio 2019 16.8.6]
 - Runtime: [e.g. .NET 4.6.2]
 - OPC Package Release Version [e.g. 1.4.365.023]
 - Component: [e.g. Ua.Quickstart]
 - Server: [e.g. Reference Server]
 - Client: [e.g. Custom Client]

Thanks in advance!

AlinMoldovean, 2021-03-03:
Hi @JayDalton ,

It seems that your Write operation is rejected at `ValidateRequest()`

```
 // reject all writes if no user provided.
                if (context.UserIdentity.TokenType == UserTokenType.Anonymous)
                {
                    // construct translation object with default text.
                    TranslationInfo info = new TranslationInfo(
                        "NoWriteAllowed",
                        "en-US",
                        "Must provide a valid windows user before calling write.");

                    // create an exception with a vendor defined sub-code.
                    throw new ServiceResultException(new ServiceResult(
                        StatusCodes.BadUserAccessDenied,
                        "NoWriteAllowed",
                        Namespaces.UserAuthentication,
                        new LocalizedText(info)));
                }

```

JayDalton, 2021-03-04:
[image]

At this it kick's me out. Because there is no difference in AccessLevel handling between different users.

So for now I do a tricky thing: Like in the "Quickstart Applications" Demo for "Boiler Server", when creating the information model, i keep the handles of my created variables in a separate Dictionary an write the values "under the hood" without library machanics.

[image]

then I write to value by handle

[image]

is this a good way to do?

AlinMoldovean, 2021-05-26:
Hello @JayDalton ,

The `WriteValueAttribute() `should also check if the OnReadUserAccessLevel handler is implemented and use the returned value.
The same for `ReadValueAttribute()`.

I will make a fix for this.

AlinMoldovean, 2021-05-26:
@JayDalton,

Can you test if the fix from #1413 solves the problem?

commit 8406716b (2021-05-31): Fixes #1302. Use value from OnReadUserAccessLevel at Read/WriteNonValueAttribute if handler exists. (#1413)
closed 2021-05-31 by AlinMoldovean