OPCFoundation/UA-.NETStandard #1303: Create custom event instances

Haeavar, 2021-03-02:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [ X ] Question [ X ] Help wanted

**Describe the Issue**

I have an custom event _myCustomAlarm_ which derives from _AlarmConditionType_. The _myCustomAlarm_ have an BaseVariable _myAlarmField_ of type string[].
How can I create an instance of this _myCustomAlarm_ with the OpcFoundation Ua Server without implementing the class having FindChild etc? If I use the _AlarmConditionState_ I will miss my _myAlarmField_ on that instance, or?

And how can I set the  _myAlarmField_ without an explicit class? Using _SetChildValue_ is not working:
`myCustomAlarm.SetChildValue(context, new QualifiedName("myAlarmField", typeDefinition.NamespaceIndex), new string[] { "hello", "world" }, false);`

mrsuciu, 2021-09-14:
Probably an AlarmAndEvents sample from the  sample repository would help

EthanChangAED, 2023-07-07:
Closing this issue. Please file a new one in the discussion-Q&A if there is more quesiton.

closed 2023-07-07 by EthanChangAED