OPCFoundation/UA-.NETStandard #1306: Support for Complex types in server (Opc.Ua.Server.ComplexTypes)

randomstring384540, 2021-03-03:
**Type of Issue**
[ ] Bug [x] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

Currently "types" are only supported on client side using Opc.Ua.Client.ComplexTypes library. It would be very helpful if ComplexTypes for nodesets that were loaded on server were also support.

For example the python OPC UA server https://github.com/FreeOpcUa/opcua-asyncio has a load_type_definitions() method which makes working with "unkown" types at runtime a breeze.

Looking at the library most things seem to be in place already, but right now it is coupled to a client session.

A clear and concise description of what the issue is.

**To Reproduce**

**Expected behavior**

**Screenshots**

**Log output**

**Code snippets**

**Environment (please complete the following information):**

**Additional context**

mregen, 2022-01-21:
Planning rewamp of complex types in the next milestone...

DerGary, 2023-11-16:
Is there an update to this feature request? We are corporate member and are desperately waiting for this feature too