OPCFoundation/UA-.NETStandard #1308: LocalTime property of AlarmConditionState is null and cannot be used

tschombe, 2021-03-04:
Hi,

I want to build an Alert and Events server. I want to use the LocalTime property of an AlarmConditionState
with the following code:
`node.LocalTime = new PropertyState<TimeZoneDataType>(this);
node.LocalTime.Value = Utils.GetTimeZoneInfo();

with these two lines, i later get an exception when constructing my address space in the final 
call:
'AddPredefinedNode(SystemContext, root);'

I get a System.ArgumentNull Exception somewhere down the callstack of AddPredefinedNode.

I cannot grasp, where my error is.

I want to write a unit test, to show the behaviour, but it is very hard to setup unittests for AlarmConditionState
and mock Context and everything correctly.

So I use AlarmConditionServer from Quickstarts to show the problem, but it worked there

In SourceState.cs where an alarm state is created, at Line 392 the ReceiveTime is set.
I append my two lines there:
`node.LocalTime = new PropertyState<TimeZoneDataType>(this);
node.LocalTime.Value = Utils.GetTimeZoneInfo();

but the Alarm ConditionServer works and does not throw the Exception in AddPredefinedNode() later on.

I am lost,
any help would be greatly appreciated.

I could offer remote session using teams or teamviewer

mrsuciu, 2021-09-14:
Is this still an issue ?

tschombe, 2021-09-20:
unfortunately i am not able to work on the project currently, so i am not sure if this is still an issue or not :-(

mregen, 2022-01-21:
Please reopen if still an issue

closed 2022-01-21 by mregen