OPCFoundation/UA-.NETStandard #1311: default  / is added by (new uri(url)  call in OPC client code

jsebes1, 2021-03-05:
**Type of Issue**
[ X] Bug [ ] Enhancement [ ] Compliance [ ] Question [ ] Help wanted

for example.

from the sample client of OPCUA foundation
[(https://github.com/OPCFoundation/UA-.NETStandard)]

In ..\UA-.NETStandard-master\Applications\ConsoleReferenceClient\UAClient.cs
keep
private string m_serverUrl = "opc.tcp://x.y.z.a:4840"

in wire shark we can see that / is added at the end by default for url "opc.tcp://x.y.z.a:4840/"

problem seems to be in

```
public static EndpointDescription SelectEndpoint(
            string discoveryUrl,
            bool useSecurity,
            int discoverTimeout
            )
        {
            **var url = GetDiscoveryUrl(discoveryUrl);**
            var endpointConfiguration = EndpointConfiguration.Create();
            endpointConfiguration.OperationTimeout = discoverTimeout;

            // Connect to the server's discovery endpoint and find the available configuration.
            using (var client = DiscoveryClient.Create(url, endpointConfiguration))
            {
                var endpoints = client.GetEndpoints(null);
                return SelectEndpoint(url, endpoints, useSecurity);
            }
        }

```
and then

```
private static Uri GetDiscoveryUrl(string discoveryUrl)
       {
           // needs to add the '/discovery' back onto non-UA TCP URLs.
           if (discoveryUrl.StartsWith(Utils.UriSchemeHttp))
           {
               if (!discoveryUrl.EndsWith("/discovery"))
               {
                   discoveryUrl += "/discovery";
               }
           }
           // parse the selected URL.
           return new Uri(discoveryUrl); //Problem here
       }
```

the call to new Uri(discoveryUrl) always returns adding a / at the end and on "some" opc servers it fails . ( Not all but in some it passes)

In case of node.js OPC the call is successful since / is not added explicitly . if we add / as part of ura there also call fails .
A clear and concise description of what the issue is.

**To Reproduce**
Steps to reproduce the behavior:
1. provide url without / at end
2. Hello Works
3.get end point works
3. create session fails due to unnecessary / at  the end on some opc servers with bad not found error
4. even if path is there the specific opc server is expecting without /
5. similar issue https://github.com/eclipse/milo/issues/208 solved in milo

**Expected behavior**
don't append  /  unnecessarily some opc servers don't handle it..

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Log output**
If applicable, add log output to help explain your problem.

**Code snippets**
If applicable, add a code snippet to help explain your problem.

Windows 10 , .net 4.6.2 ( any version , or latest )
using opc stack latest version and sample client

AlinMoldovean, 2021-03-15:
Hello @jsebes1 ,

The trailing slash is added by the `Uri `class of .NET. Framework.
I other cases this was handled on server side.
See [.NET-Legacy/ Issue #35](https://github.com/OPCFoundation/UA-.NET-Legacy/issues/35) 
or your reference [eclipse/milo#208](https://github.com/eclipse/milo/issues/208)

jsebes1, 2021-03-15:
The problem is this specific server is not accepting the  /  and works without /  , The server is not based on .Net framework or Milo framework . Probably I can override some methods or doing some dirty coding can achieve this . But if framework was providing so , it was fine. 

UaExpert  client and nodeJS opc client works fine with this server since they are able to send the request without /
So any mechanism in core stack to send without / would have been helpful.

AlinMoldovean, 2021-06-08:
Hello @jsebes1 ,

I made a test again with the latest version of .NET Framework 4.6.2 (latest patches applied), .NET Core.App 2.1 and 3.1.
It seems that the `Uri.ToString()` implementation does not add the trailing slash anymore, so the issue should be fixed.

[image]

[image]

Could you check this?

mrsuciu, 2021-09-15:
Should be fixed

closed 2021-09-15 by mrsuciu