OPCFoundation/UA-.NETStandard #1312: Browsing through a Nodeset (with xml file)

Yoann85, 2021-03-05:
**Type of Issue**
[ ] Bug [ ] Enhancement [ ] Compliance [ ] Question [X] Help wanted

Hi Everyone,
I would like to configure an OPC UA client offline, using the xml nodeset exported from a server.
For now, I obtained a list of nodes :

```C#
UANodeSet nodeSet = UANodeSet.Read(file);
UANode[] NodeList= nodeSet.Items;
```

I'm trying to loop on this list to construct a hierarchical tree that describes the server's address space content. But I'm unable to do it the same way as in online mode.
If I get it, `UANode` has sub-types such as `UAObject`, `UADataType`, `UAVariable`, etc...
Some sub-types contain a `ParentNodeId` property but each `UANode` must be casted into a compatible type to access it, which is not convenient...

Which UANode's property should I use to link a node to its parent? How can I find the root Node?
I'm far from being an expert, is there a better way to browse offline?

Thank you

Yoann85, 2021-03-14:
Hi,

Maybe should I try to cast each node as [UAInstance](https://reference.opcfoundation.org/v104/Core/docs/Part6/F.7/) and skip all the nodes that are not convertible. Then use the ParentNodeId to build the hierarchical tree. That way I would only lose non-variable nodes according to [UANodeSet Reference](https://reference.opcfoundation.org/v104/Core/docs/Part6/F.2/).

Doesn't seem to be a clean way to do it but I can't see anything else for now.
If someone has a clue, he would be welcome!

AlinMoldovean, 2021-03-16:
Hi @Yoann85 ,

You can cast the nodes as NodeState instances and NodeStateCollection and do a similar Browse implementation as the the one from server.

https://github.com/OPCFoundation/UA-.NETStandard/blob/d7236cebc1860a55fea9be7612381755e8656bbb/Stack/Opc.Ua.Core/Schema/UANodeSetHelpers.cs#L113-L121

https://github.com/OPCFoundation/UA-.NETStandard/blob/d7236cebc1860a55fea9be7612381755e8656bbb/Stack/Opc.Ua.Core/Schema/UANodeSetHelpers.cs#L451

closed 2021-03-31 by AlinMoldovean

Yoann85, 2021-03-31:
Thank you for you Help Alin, I will test it ASAP